prusa3d/PrusaSlicer #2228: [Feature request] 0.25mm profiles for MK3S/MMU2S

sepotvin, 2019-05-14:
Would it be possible to add support for 0.25mm nozzle profiles with MMU2S or is there an intrinsic limitation to 0.4mm nozzles?

bubnikv, 2019-05-14:
We probably could, but it is non-trivial to print with such a diameter, you
will probably better off to buy the SLA printer from us.

The E3D have a write up on how to print with a 0.25mm nozzle and how
difficult it is. They were able to print just ABS.

sepotvin, 2019-05-14:
I get relatively good results in single mode so far, but you're right, it's much less forgiving. What about adding support for single mode MMU2 only then? This should be much easier to implement than changing the purge tower code.

sepotvin, 2019-05-14:
Thanks a lot!!! It looks like my wish is already taken care of in the latest 2.0.0-rc version :) Talk about good service!

bubnikv, 2019-05-15:
Could we close this issue then?

sepotvin, 2019-05-15:
Yes, thanks again

closed 2019-05-15 by sepotvin