prusa3d/PrusaSlicer #2229: If stl has period in name, generated gcode file has incorrect name when sending to OctoPrint

jimmy00784, 2019-05-14:
### Version
1.42.beta2

### Operating system type + version
Archlinux 64 bit

### 3D printer brand / version + firmware version (if known)
CoreXY, Klipper

### Behavior
1. Configure the printer profile for OctoPrint
2. Slice a file with a period in name (eg RAMPS_Case_Left_v1.6.stl)
3. Click "Send G-Code" to send to OctoPrint and observe the name
4. Click "Ok"

OctoPrint will throw an error saying couldn't upload due to invalid type.

5. Fix the name to have an ".gcode" extension.

OctoPrint will accept the upload.

[image]
[image]
[image]

reloxx13, 2019-05-14:
confirm, this is so annoying >.<
many designers on thingy add a dot in the filename like xyz_v1.2.stl or xyz_2.5mm.stl

BerndJM, 2019-05-14:
Could be fixed by adding .gcode in
Print Settings -> Output Options -> Output filename format
so that it reads: `[input_filename_base].gcode`

jimmy00784, 2019-05-15:
@BerndJM Thanks. That fixes it in the interim. But it'd still be only a workaround as the default behavior causes the issue and will need to be either manually fixed at the time of sending gcode or fixed in every profile used with OctoPrint and possibly other Wifi based systems.

bubnikv, 2019-05-15:
It seems to be working in PrusaSlicer 2.0.0-rc. Would you please retest? Thanks.

reloxx13, 2019-05-15:
@bubnikv 
Unfortunately not

[image]

bubnikv, 2019-05-15:
what exactly is the problem? Would you please be more explicit? Is it that
the .gcode extension is missing in the upload dialog? Or something else?

jimmy00784, 2019-05-15:
@bubnikv If the stl file name has a period in it, then the file name in the Send GCode dialog is missing the .gcode extension.
sliced.stl gets converted correctly to sliced.gcode
but sliced.v1.stl gets converted to sliced.v1 which octopi rejects as an invalid file type.

bubnikv, 2019-05-15:
ok, that is a known issue. You need to add the .gcode extension to the end of your output filename template.

This limitation is there to support custom G-code extensions. Anything after the last dot is considered to be a custom file extension.

bubnikv, 2019-05-15:
This issue is a duplicate of https://github.com/prusa3d/PrusaSlicer/issues/1918
Closing this one.

closed 2019-05-15 by bubnikv

jimmy00784, 2019-05-15:
> ok, that is a known issue. You need to add the .gcode extension to the end of your output filename template.
> 
> This limitation is there to support custom G-code extensions. Anything after the last dot is considered to be a custom file extension.

I am not sure I follow. In both instances, stl was after the the last dot, but it works in one case and doesn't in another.
I also tested sliced.v.1.stl and it gets converted to sliced.v.1 
Seems to me that there might be an indexing issue.

jimmy00784, 2019-05-15:
Here is the issue 
Slic3r/src/libslic3r/PrintBase.cpp:61
```
// Generate an output file name based on the format template, default extension, and template parameters
// (timestamps, object placeholders derived from the model, current placeholder prameters, print statistics - config_override)
std::string PrintBase::output_filename(const std::string &format, const std::string &default_ext, const DynamicConfig *config_override) const
{
    DynamicConfig cfg;
    if (config_override != nullptr)
    	cfg = *config_override;
    PlaceholderParser::update_timestamp(cfg);
    this->update_object_placeholders(cfg);
    try {
        boost::filesystem::path filename = this->placeholder_parser().process(format, 0, &cfg);
        if (filename.extension().empty())
        	filename = boost::filesystem::change_extension(filename, default_ext);
        return filename.string();
    } catch (std::runtime_error &err) {
        throw std::runtime_error(L("Failed processing of the output_filename_format template.") + "\n" + err.what());
    }
}
```
The logic check `if(filename.extension().empty()` is inadequate. 
See its documentation [here](https://www.boost.org/doc/libs/1_45_0/libs/filesystem/v3/doc/reference.html#path-extension)
At least check if the extension matches a white list.
This needs to be re-opened

reloxx13, 2019-05-15:
stop posting in a duplicated issue!!!