prusa3d/PrusaSlicer #2232: filament type (variable string) not saved

akust, 2019-05-14:
### Version
_Version of PrusaSlicer used goes here_
Slic3rPE-1.42.0-beta2+

### Operating system type + version
macOS Sierra 10.12.6

_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
i3 mk3

### Behavior
* _Describe the problem_
you can enter a free string in Filament Settings/Advanced/Filament Type. But it is forgotten but not saved in the preset/template nor in the export. In my example case of BVOH it's a problem because the default setting here is set to "PLA". But it's not PLA at all (Palette splicing needs very different settings). 
* _Steps needed to reproduce the problem_
- in Filament Settings call the template "Verbantim BVOH"
- go to Advanced/Filament Type (set to "PLA"): replace it by writing BVOH
- save template as a user saved one.
- call a different template and go back to this just saved one
* _Expected Results_
- settings as saved,  Advanced/Filament Type set to "BVOH"

* _Actual Results_
Advanced/Filament Type is again "PLA"

_Is this a new feature request?_
nope, but a bug

YuSanka, 2019-05-15:
Thanks for your report. I can confirm it. Working on it in [JIRA SPE-933](http://dev.prusa3d.com/browse/SPE-933)

FidelCapo, 2019-05-15:
This issue should be fixed in next release.
Closing.

closed 2019-05-15 by FidelCapo