prusa3d/PrusaSlicer #2233: PrusaSlicer 2.0 crash systematically on macOs

BastR, 2019-05-14:
### Version
V2.0rc

### Operating system type + version
macOs 10.14.4 (18E226)

### Behavior
**First try:**
-Fresh install
-Imported config bundle from previous version
-When I load a STL file, app crash.

**Second try:**
-Fresh install
-Delete all previous user profiles with the configuration assistant
-Create a new 3D printer as custom printer
-When I load a STL file, app crash..

**Third try:**
-Fresh install
-Delete all previous user profiles with the configuration assistant
-Set 3D printer as MK3
-When I load a STL file, app crash..

I attach complete crash log
[Crashlog.txt](https://github.com/prusa3d/PrusaSlicer/files/3179846/Crashlog.txt)

Raekro, 2019-05-14:
Same issue.

Version

V2.0rc

Operating system type + version

macOs 10.14.4

BastR, 2019-05-14:
Ok just an update:

**Fourth try:**
-Set app language to English, the language of the application was the same as the system (French)
-When I load a STL file, app work !!!
-Then set app language to French and app crash.

So I don't if @Raekro is French, but this is maybe a problem with French translation

tazlewouaf, 2019-05-14:
Hi,
I'm facing the same issue with French language on Windows x64 with "PrusaSlicer-2.0.0-rc+win64-201905141708.zip"
App doesn't crash in English :)

akust, 2019-05-14:
I found it was auto local (german), set it to english - crash.
[slic3r first start language crashlog.txt](https://github.com/prusa3d/PrusaSlicer/files/3179980/slic3r.first.start.language.crashlog.txt)

Raekro, 2019-05-14:
> Ok just an update:
>
> **Fourth try:**
> -Set app language to English, the language of the application was the same as the system (French)
> -When I load a STL file, app work !!!
> -Then set app language to [...]

I am French, but I usually use Slic3r in English. I was surprised to see him in French at the first launch. Indeed, after changing the language, everything is ok.

shro70, 2019-05-14:
Same in win7 64bits

FidelCapo, 2019-05-15:
Thank you for report. I can reproduce it.

Created JIRA-932

jpatenet, 2019-05-15:
Same issue, app crash in french but works well in english

FidelCapo, 2019-05-16:
This issue should be fixed in next release.
Closing

closed 2019-05-16 by FidelCapo