prusa3d/PrusaSlicer #2235: 2.0.0rc MacOSx dark mode illegible

maxcom99, 2019-05-14:
### Version
ersion_2.0.0-rc-1-g3f08d9f30

### Operating system type + version
OSx Mojave
### 3D printer brand / version + firmware version (if known)
N/A
### Behavior
Starting the app with dark mode enabled on MacOS makes the text illegible.

_Is this a new feature request?_
No
[image]

FidelCapo, 2019-05-15:
[image]

Can you please send me screenshot of this settings from your mac? Thank you.

bubnikv, 2019-05-15:
Did you possibly install any 3rd party application to tweak the user
interface colors or theme? Do you remember setting anything else apart from
the dialog above to tweak the appearance?

bubnikv, 2019-05-15:
What exactly is your OSX Mojave version? Are you on the latest update? What
is your machine type?

bubnikv, 2019-05-15:
By the way, we are running PrusaSlicer on multiple computers with OSX
Mojave and we never experienced your issue.

liftbag, 2019-05-16:
I'm using dark mode from Mojave day one and I've never experienced this issue too.

maxcom99, 2019-05-16:
[image]

maxcom99, 2019-05-16:
[image]

maxcom99, 2019-05-16:
Looks like the issue is only with the self compiled, the released version works. Closing.

closed 2019-05-16 by maxcom99

vojtechkral, 2019-05-17:
I've seen this problem before. I don't remember 100% what was the cause, but I _think_ it was due to wxwidgets being compiled against a Mojave SDK but with older OSX deployment target. As far as I know & remember, both the target and sdk version must match across slicer, wxwidgets and the other deps, otherwise it either won't compile or you'll get issues like these...