prusa3d/PrusaSlicer #2236: Unable to import config bundle

uacajala, 2019-05-14:
Version: 2.0.0-rc+
Build: PrusaSlicer-2.0.0-rc+-201905141706
Operating System: Macintosh
System Architecture: 64 bit
System Version: macOS Version 10.14.4 (Build 18E226

### 3D printer brand / version + firmware version (if known)
N/A

### Behavior
Tried to import config bundle exported from previous main slicer:

Version: 1.42.0-beta2+
Build: Slic3rPE-1.42.0-beta2+full-201904140836
Operating System: Macintosh
System Architecture: 64 bit
System Version: macOS Version 10.14.4 (Build 18E226)

Previously had no issues with the old slicer (Slic3r PE 1.42.0 beta2). I exported the config bundle from the slicer above successfully. I could not import the config bundle into the new slicer (PrusaSlicer 2.0.0 rc2) which the issue is regarding (details at the top). 

To recreate the steps carry out the steps above. I have also attached the ini file that was used for the config import that failed after multiple attempts.

[Slic3r_config_bundle.ini.zip](https://github.com/prusa3d/PrusaSlicer/files/3180208/Slic3r_config_bundle.ini.zip)

bubnikv, 2019-05-15:
The square brackets are not allowed in the profile names. We shall change
the application to test for them.

For now, please edit your config bundle file manually, and replace all
occurrences of [Volcano] with something without square brackets. You will
then be able to import the file.

bubnikv, 2019-05-15:
The square brackets are now banned in the profile name
44306e17e93f3120f0b781abef83256c3f3335d0

Closing.

closed 2019-05-15 by bubnikv