prusa3d/PrusaSlicer #2238: PrusaSlicer 2.0.0-rc - Bug Report

3Delights, 2019-05-15:
### Version
PrusaSlicer 2.0.0-rc

### Operating system type + version
Windows 10 Pro x64 Fully updated.

### 3D printer brand / version + firmware version (if known)
MK2 (No S).

### Behavior
Every time I run it, it pops the error message "Couldn't add an image to the image list." The details drop down contains:

13:10:05: Couldn't add an image to the image list.
13:10:05: Couldn't add an image to the image list.
13:10:05: Couldn't add an image to the image list.
13:10:05: Couldn't add an image to the image list.
13:10:05: Couldn't add an image to the image list.
13:10:05: Couldn't add an image to the image list.
13:10:05: Couldn't add an image to the image list.
13:10:05: Couldn't add an image to the image list.
13:10:05: Couldn't add an image to the image list.
13:10:05: Couldn't add an image to the image list.
13:10:05: Couldn't add an image to the image list.
13:10:05: Couldn't add an image to the image list.
13:10:05: Couldn't add an image to the image list.
13:10:06: Couldn't add an image to the image list.
13:10:06: Couldn't add an image to the image list.
13:10:06: Couldn't add an image to the image list.
13:10:06: Couldn't add an image to the image list.
13:10:06: Couldn't add an image to the image list.
13:10:06: Couldn't add an image to the image list.
13:10:06: Couldn't add an image to the image list.
13:10:06: Couldn't add an image to the image list.
13:10:06: Couldn't add an image to the image list.
13:10:06: Couldn't add an image to the image list.
13:10:06: Couldn't add an image to the image list.
13:10:06: Couldn't add an image to the image list.
13:10:06: Couldn't add an image to the image list.
13:10:06: Couldn't add an image to the image list.
13:10:06: Couldn't add an image to the image list.
13:10:06: Couldn't add an image to the image list.

Clicking OK allows it to run as normal, no idea what images (if any) are missing!?!

YuSanka, 2019-05-15:
Thanks for your report. Do you have 2 displays? Can you see icons in Options List on left panel in Settings Tabs?

pritch55, 2019-05-15:
I'm getting the same error, it looks like it might be the image of the bed. First picture is Slic3r and the second picture is PrusaSlicer.

[image]

[image]

bubnikv, 2019-05-15:
@3Delights  a screenshot is worth thousand words. Thank you.

pritch55, 2019-05-15:
I'm getting the same error, here are a couple screenshots.

[image]
[image]
[image]

YuSanka, 2019-05-16:
Could you try this build, please, 
[PrusaSlicer-2.0.0-rc+16-ys-bugfixing-win64-g0bcb449fb-201905151450.zip](https://github.com/prusa3d/PrusaSlicer/files/3185676/PrusaSlicer-2.0.0-rc%2B16-ys-bugfixing-win64-g0bcb449fb-201905151450.zip)

bubnikv, 2019-05-16:
Let me comment on each of the issues separately. As usual, there are different issues mixed, which manifest in a similar way.

@3Delights and @pritch55 seem to have an issue, where Windows report a font size (by the Win32 SystemParametersInfo(NONCLIENTMETRICS) function, parameter lfMessageFont), which is non-standard, likely slightly larger than usual. This is manifested by the application font being slightly larger than the menu or title bar font. @YuSanka posted a build above, which fixed a similar issue on Linux, hopefully it will fix your issue as well. Please help to test it, thanks.

We still do not know why your fonts are larger than usual. Maybe you used some 3rd party application like https://betanews.com/2018/03/14/roundup-windows-10-tweaking-customization/ to adjust the font sizes? 

@YuSanka saw your issue once and we have had no other report in house. The PrusaSlicer is used likely on fifty to hunderd of computers, so this issue must be pretty rare. Likely this issue happened to @YuSanka after she switched resolution of one of her displays, but she is not able to reproduce it anymore. I just wonder how we can reproduce such an issue in the future.

@pritch55's 2nd screenshot shows slightly bigger fonts, but I would say the application is perfectly usable. Only the icons in the left panel are missing. On the 3rd screenshot everything is fine. The 4rd screenshot shows an old error (Slic3r did not work correctly if installed in a path with non-Latin1 characters) and we hope that the latest PrusaSlicer will work there without a problem.

pritch55, 2019-05-16:
The build YuSanka provided worked, thanks for your help.

danielle6192, 2019-05-21:
hey i have a problem with all the icones they dosent work and i cant see the estimated printing time aswell so i cant use the slicer

bubnikv, 2019-05-22:
> hey i have a problem with all the icones they dosent work and i cant see
the estimated printing time aswell so i cant use the slicer

Sorry, I don't undestand. Would you please be more specific?

danielle6192, 2019-05-22:
yeah the new slicer dosent work well// i cant prass the icons on the top
and the left side and the estimated time is also incorect

‫בתאריך יום ד׳, 22 במאי 2019 ב-10:53 מאת ‪bubnikv‬‏ <‪
notifications@github.com‬‏>:‬

bubnikv, 2019-05-22:
the time estimate depends on how precisely the slicer could simulate your firmware. You have to set the machine limits correctly.

Also I have no idea why you cannot press the icons. Would you show a screenshot or a video?