prusa3d/PrusaSlicer #2240: invalid bitmap size in wxImageList at startup

jasontibbitts, 2019-05-15:
### Version
2.0.0 rc

### Operating system type + version
Linux Fedora 29 x86_64 using self-compiled packages, using wxGTK3 3.0.4.

### Behavior
At startup I get a number of dialogs saying:

An assertion failed!
../src/generic/imaglist.cpp(66): assert "(bitmap.GetWidth() >= m_width && bitmap.GetHeight() == m_height) || (m_width == 0 && m_height == 0)" failed in Add(): invalid bitmap size in wxImageList: this might work on this platform but definitely won't under Windows.

ASSERT INFO:
../src/generic/imaglist.cpp(66): assert "(bitmap.GetWidth() >= m_width && bitmap.GetHeight() == m_height) || (m_width == 0 && m_height == 0)" failed in Add(): invalid bitmap size in wxImageList: this might work on this platform but definitely won't under Windows.

BACKTRACE:
[1] wxGenericImageList::Add(wxBitmap const&)
[2] Slic3r::GUI::Tab::add_options_page(wxString const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, bool)
[3] Slic3r::GUI::TabPrint::build()
[4] Slic3r::GUI::Tab::create_preset_tab()
[5] Slic3r::GUI::MainFrame::add_created_tab(Slic3r::GUI::Tab*)
[6] Slic3r::GUI::MainFrame::create_preset_tabs()
[7] Slic3r::GUI::MainFrame::init_tabpanel()
[8] Slic3r::GUI::MainFrame::MainFrame()
[9] Slic3r::GUI::GUI_App::on_init_inner()
[10] Slic3r::GUI::GUI_App::OnInit()
[11] wxEntry(int&, wchar_t**)
[12] Slic3r::CLI::run(int, char**)
[13] main
[14] __libc_start_main
[15] _start

If I click through them all (or uncheck "Show this dialog the next time" and continue), the program starts and appears to work fine. I believe the dialog appears 29 times at startup.

It occurs to me that this seems similar to #2238, where there are complaints about 29 images at startup.  Though that is on Windows, where (if the message is to be believed) the operation fails completely.

YuSanka, 2019-05-16:
Thanks for your report. Those assertions are fixed. You can try [ys_bugfixin](https://github.com/prusa3d/PrusaSlicer/commits/ys_bugfixing) or wait for new release.

bubnikv, 2019-05-16:
Fixed, closing.

closed 2019-05-16 by bubnikv