prusa3d/PrusaSlicer #2241: [Feature request] SLA horizontal/vertical mirror output

bobtiki, 2019-05-15:
### Version

2.0.0-rc

### Operating system type + version

macOS Mojave 10.14.4 (18E226)

### 3D printer brand / version + firmware version (if known)

Wanhao D7, stock model

### Behavior

- This is a new feature request.
- When exporting the slice raster images, the images exported for the Prusa SL1 are flopped (horizontally mirrored) from the images the Wanhao D7 expects to receive, leading to “inside out” prints, if I upload these images files through NanoDLP for printing.
- **Proposal:** add horizontal/vertical mirror controls to the Printer Settings dialog, to assign a mirror setting for an SLA printer, and export accordingly.

[image]

In the meantime, I have written a small Python script that flops and repackages the images: https://github.com/bobtiki/flop_3d_layers

bubnikv, 2019-05-16:
That should be trivial to add, but I am not sure whether we will manage for the PrusaSlicer 2.0.0 release.
@tamasmeszaros will look into it.

bubnikv, 2019-08-01:
This will be implemented by the upcoming PrusaSlicer 2.1.0-alpha1.
Closing.

closed 2019-08-01 by bubnikv

Henke1983, 2020-01-01:
> ### Version
> 2.0.0-rc
>
> ### Operating system type + version
> macOS Mojave 10.14.4 (18E226)
>
> ### 3D printer brand / version + firmware version (if known)
> Wanhao D7, stock model
>
> ### Behavior
> * This is a [...]

Hi @bobtiki !
I saw your post about Wanhao D7. I tried to do a profile for Wanhao D7 in PrusaSlic3r and it ended with a SL1 file - should I just rename it to what WanahoD7 wants or how have you done it? Is it also possible to share your settings with me ?

Great new year and Thank you in advance! :)