prusa3d/PrusaSlicer #2243: Do not put __FILE__ info in release mode

Salamandar, 2019-05-15:
The macro __FILE__ is often used in the code. That may be a good idea to **not** use it in release mode, only in debug mode. 

```
 ❯ grep __FILE__ src/PrusaSlicer/src -R
src/PrusaSlicer/src/igl/ply.h:#define myalloc(mem_size) my_alloc((mem_size), __LINE__, __FILE__)
src/PrusaSlicer/src/igl/ply.h:		           __LINE__, __FILE__);                             		\
src/PrusaSlicer/src/igl/ply.h:		 __LINE__, __FILE__);                           \
src/PrusaSlicer/src/avrdude/avrftdi_private.h:					__FILE__, __LINE__, __FUNCTION__,                         \
src/PrusaSlicer/src/avrdude/avrftdi_private.h:					__FILE__, __LINE__, __FUNCTION__,                         \
src/PrusaSlicer/src/eigen/Eigen/src/Core/util/Macros.h:          Eigen::internal::assert_fail(EIGEN_MAKESTRING(x), __PRETTY_FUNCTION__, __FILE__, __LINE__); \
src/PrusaSlicer/src/slic3r/GUI/3DScene.hpp:inline void glAssertRecentCall() { glAssertRecentCallImpl(__FILE__, __LINE__, __FUNCTION__); }
src/PrusaSlicer/src/slic3r/GUI/3DScene.hpp:#define glsafe(cmd) do { cmd; glAssertRecentCallImpl(__FILE__, __LINE__, __FUNCTION__); } while (false)
src/PrusaSlicer/src/slic3r/GUI/3DScene.hpp:#define glcheck() do { glAssertRecentCallImpl(__FILE__, __LINE__, __FUNCTION__); } while (false)
```

bubnikv, 2019-05-16:
igl, avrdude and eigen are external dependencies, I don't think it is wise
to change them.
With the exception to avrdude, I am pretty sure that the __FILE__, __LINE__
etc are only compiled in debug mode, which is what we want.
IMHO it is all fine.
Closing.

closed 2019-05-16 by bubnikv

Salamandar, 2019-05-19:
I confirm that it's also used in Release mode. I'm the maintainer of the AUR prusaslicer-git package and had warnings about this.

Of course I wasn't thinking about the dependencies ;) Only about the `slic3r/GUI/*` files.

bubnikv, 2019-05-19:
how did you confirm that it is used in release mode?

Salamandar, 2019-05-19:
1. By reading the code : There are no #define that conditionally include this code. It's always included and called.
2. I actually was tracking WHY the binaries were still containing a string with the build directory path in it. In the release binary, with release CFLAGS and CMake flags. And that's why.

Salamandar, 2019-05-19:
The code probably should be : 
```
# src/PrusaSlicer/src/slic3r/GUI/3DScene.hpp:
#ifdef DEBUG
#define glsafe(cmd) do { cmd; glAssertRecentCallImpl(__FILE__, __LINE__, __FUNCTION__); } while (false)
#else
#define glsafe(cmd) do { cmd; } while (false)
#endif
```
The DEBUG macro probably has another name in PrusaSlicer.

bubnikv, 2019-05-20:
#ifndef NDEBUG
#define HAS_GLSAFE
#endif

NDEBUG is defined in the release builds

#ifdef HAS_GLSAFE
all the code that offends you is here
#endif

Salamandar, 2021-11-12:
Well that issue is again relevant.
https://github.com/prusa3d/PrusaSlicer/blob/master/src/slic3r/GUI/3DScene.hpp#L16