prusa3d/PrusaSlicer #2244: Spiral Vase throws a multi-material error

devcodeio, 2019-05-16:
### Version
2.0.0-rc + win64

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
Prusa MK3S - Recent firmware

### Behavior
* Loaded a 3D Model 
[www.thingiverse.com/thing:2376777](url)
* Went into print settings and selected _Spiral Vase_ 
* Expected it to slice successfully
* Received a multi-material warning, and _invalid data_ was displayed
[image]

### Notes
Sliced without issue using the following version:
Slic3rPE 1.42.0 + beta2 + win64

bubnikv, 2019-05-16:
I cannot reproduce it with the  2.0.0-rc + win64. Would you please attach
your project file?

devcodeio, 2019-05-16:
I replicated it many times yesterday, despite there being only one material i would still be given that notification.  I loaded up the project today and now things are slicing without a hitch?    
Ill close and if i come upon it again ill reopen it

closed 2019-05-16 by devcodeio