prusa3d/PrusaSlicer #2245: latest slicer - supports create overkill material that actually sopports nothing above it.

SirBastol, 2019-05-16:
### Version
_Version of PrusaSlicer used goes here_

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_

SirBastol, 2019-05-16:
slicer pe makes supports that support nothign above it if enabled wasting filament and time.. 
enable tree supports or vertical checks to make sure the support is supporting something..

bubnikv, 2019-05-16:
Please provide screenshots and the PrusaSlicer project file. Thank you.

Axeman72, 2019-05-19:
(Not the OP)
It happens often to me that slic3r add supports with nothing over them. Sometimes they are "real" support, sometimes they're only on first layer, like in this case: 
[image]
.3mf file attached.

[Knob.zip](https://github.com/prusa3d/PrusaSlicer/files/3195350/Knob.zip)

megamih, 2019-05-23:
### Version
PrusaSlicer-2.0.0+win64-201905201652

### Operating system type + version
Microsoft Windows 10 Home 64bit 1809 Build 17763

### 3D printer brand / version + firmware version (if known)
NA

### Behavior
* Automatic supports in "Everywhere" mode create overkill material that actually sopports nothing above it

* Expected Results

[image]

*Actual Results

[image]

#### Project File (.3MF) where problem occurs

[stk1ax32sc_holder_v3.zip](https://github.com/prusa3d/PrusaSlicer/files/3214224/stk1ax32sc_holder_v3.zip)

nordurljosahvida, 2019-11-16:
Happens to me all of the time as well. I would have +1 commented an above comment but i see this issue has been dormant for a while. Thanks

bubnikv, 2021-02-24:
The original issue is fixed with 539e80e

When expanding the 1st layer support layer, do it by small steps while
trimming with object to avoid supports to leak through object walls.

closed 2021-02-24 by bubnikv