prusa3d/PrusaSlicer #2247: PrusaSlicer-rc: in-window buttons are now enormous

Mirarkitty, 2019-05-16:
### Version
PrusaSlicer-rc AppImage under LInux, retina (4k) display

### Behavior
The overlayed buttons on the plater are now ENORMOUS. Are they using raw dpi for sizing instead of using a text size for sizing? They were good in last Sli3rPE.

Below is screenshot of PrusaSlicer vs Slic3rPE.

[image]

Mirarkitty, 2019-05-16:
If they are designed to be ENORMOUS then I guess just close this ticket. Thought you might want to know about the change if it was unintended.

guestisp, 2019-05-16:
I can confirm. I've thought it was expected

bubnikv, 2019-05-16:
They are scaled based on your font size, they should be of the same size on all platforms. I believe it works as intended.

I understand your sentiment, the icons may seem huge to some. It works as intended. In the future versions we plan to make them user configurable, and we plan to scale them down on physcally small displays, but we will most likely not make it in the PrusaSlicer 2.0.0 final release.

Works as intended, closing.

closed 2019-05-16 by bubnikv

Mirarkitty, 2019-05-16:
It seems very large but ok. Maybe someone can post a screenshot from a non-HiDPI to compare? 
[image]

guestisp, 2019-05-16:
1920x1200 on a 22''. I don't think it's HiDPI

[image]

bubnikv, 2019-05-16:
the icons scale by the application font size. That is clearly visible from
the two last screenshots. It works as intended.

Mirarkitty, 2019-05-16:
Yep

tedder, 2019-05-18:
Here's a comparison showing the interface with "fonts 200%" and "fonts 100%" in ubuntu/gnome on a 4k. It may be 'works as intended' but it's not how other apps behave.

[image]
[image]

The windows are the same size. It's an interesting design decision, I guess.

bubnikv, 2019-05-19:
> It may be 'works as intended' but it's not how other apps behave.

Because they do not really support the concept of display resolution.

We are currently linking against GTK2 on our build server. GTK3 is supposed
to support the concept of display scaling, we shall look into it in the
future together with separate GTK2 / GTK3 PrusaSlicer builds (or we may
just migrate everything to GTK3). For now, scaling buttons by the font size
seems logical at to me and our group.

> It's an interesting design decision, I guess.

It is a decision dictated by the new crop of the high resolution displays
and the necessity to support them.

Mirarkitty, 2019-05-19:
**I appreciate the scaling very much.** I'm just surprised about GUI buttons that are about twice as large as any other button in any other program, but I guess that's the size they always were for everyone that didn't have HiDPI.

Compared to for instance Gimp, Inkscape or Photoshop, this is actually useful.