prusa3d/PrusaSlicer #2248: Show text below the GUI button icons

MikePfaff, 2019-05-16:
### Version
2.0.0-rc

### Operating system type + version
Any

### 3D printer brand / version + firmware version (if known)
Any

### Behavior
It would be nice if the GUI button icons had some text (and maybe also the keyboard shortcut) shown below them describing what each button does. This would make it easier for new (or upgrading) users to understand what each of the buttons does instead of having to hover over every button and wait for the tooltip to see what the button does.

This is a new feature request.

#### Project File (.3MF) where problem occurs
Any

bubnikv, 2019-05-16:
I personally would only do it if there are multiple requests for it.

faboaic, 2019-05-18:
I am sure that this would help.
I used every version of the new generation but still have to think sometimes what which symbol means.
Especially the symbols on top of the plater.

git-prusaslicer, 2019-05-25:
[icons-text.tar.gz](https://github.com/prusa3d/PrusaSlicer/files/3219817/icons-text.tar.gz)

Made some quick-n-dirty changes to impotent icons to add text.  Makes the gui a lot more usable!  Just untar it into `resources/icons` directory and you get some basic identifiers for the main buttons.  Did not do them all, just the ones the main interface has by default.

Hope it helps someone!

larsrc, 2020-03-02:
Came here to +1. I didn't realize there was a tooltip, though, as it takes ~3s to appear, which is a long time for a UI reaction. Frankly, I would prefer just having text - even Gmail accepted that text buttons were more useful than icons.

bubnikv, 2020-03-23:
Just released PrusaSlicer 2.2.0 improves reliability of the tooltips. They are now painted directly into the 3D scene and they pop up reliably on all platforms.

rtyr, 2021-01-28:
I believe this is solved since 2.2.0. Closing.

git-prusaslicer, 2021-02-24:
Its really slow to go to every single button and hover to figure out what it does.  The original ask was to get the tooltip text under the button so it was always visible and easy to identify what button does what.  As such its not really fixed in any respect by modifying the tooltips.  I am still untaring my ugly little icon hack with every release to know what the buttons do, would be really nice if there was an option to show text next to all buttons.

closed 2026-09-29 by prusaslicer-bot[bot]