prusa3d/PrusaSlicer #2249: Colorprint: the pause insertion symbol (+) disappears after the first use.

liftbag, 2019-05-16:
### Version
_PrusaSlicer-2.0.0-rc+_

### Operating system type + version
_macOS Mojave 10.14.4_

### Behavior
* _Sometimes, the plus symbol, necessary for the colorprint function, disappears after the first use. In its place appears the X symbol, which complicates the subsequent management of the changing the filament pauses _
* _To trigger the problem, just insert a pause at a certain Z height. Most likely the + symbol will become an X and will no longer change. Sometimes this happens after entering or removing some pauses._
* _As in the SlicerPE-1.42.0-beta2 version, I expect the X symbol to appear only in correspondence with the layer affected by the pause._
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

[image]

bubnikv, 2019-05-16:
I believe it works as intended. The cross indicates, that if you are over
an existing tick, you will delete it with successive clicking on the cross.

bubnikv, 2019-05-16:
Thanks for your report, we were able to reproduce it.

YuSanka, 2019-05-16:
Bug is fixed and will be a part of next release

closed 2019-05-16 by YuSanka