prusa3d/PrusaSlicer #2250: New assertion failure with 2.0.0rc

ArchangeGabriel, 2019-05-16:
### Version
2.0.0rc

### Operating system type + version
ArchLinux

### Behavior
Launching `prusa-slicer` results in:
```
./src/generic/imaglist.cpp(66): assert "(bitmap.GetWidth() >= m_width && bitmap.GetHeight() == m_height) || (m_width == 0 && m_height == 0)" failed in Add(): invalid bitmap size in wxImageList: this might work on this platform but definitely won't under Windows.
```
Backtrace from the same window:
```
ASSERT INFO:
./src/generic/imaglist.cpp(66): assert "(bitmap.GetWidth() >= m_width && bitmap.GetHeight() == m_height) || (m_width == 0 && m_height == 0)" failed in Add(): invalid bitmap size in wxImageList: this might work on this platform but definitely won't under Windows.

BACKTRACE:
[1] wxGenericImageList::Add(wxBitmap const&)
[2] Slic3r::GUI::Tab::add_options_page(wxString const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, bool)
[3] Slic3r::GUI::TabPrint::build()
[4] Slic3r::GUI::Tab::create_preset_tab()
[5] Slic3r::GUI::MainFrame::add_created_tab(Slic3r::GUI::Tab*)
[6] Slic3r::GUI::MainFrame::create_preset_tabs()
[7] Slic3r::GUI::MainFrame::init_tabpanel()
[8] Slic3r::GUI::MainFrame::MainFrame()
[9] Slic3r::GUI::GUI_App::on_init_inner()
[10] Slic3r::GUI::GUI_App::OnInit()
[11] wxEntry(int&, wchar_t**)
[12] Slic3r::CLI::run(int, char**)
[13] main
[14] __libc_start_main
[15] _start
```

ArchangeGabriel, 2019-05-16:
Note that this issue cannot be skipped, it just happens everytime you click on “Continue”.

ArchangeGabriel, 2019-05-16:
Seems a duplicate of https://github.com/prusa3d/PrusaSlicer/issues/2240 actually. Is there a specific commit for me to try?

YuSanka, 2019-05-16:
:) you can try last master commit. It should be fixed

ArchangeGabriel, 2019-05-16:
Indeed fixed with this commit. However there is another one, when going to the “About PrusaSlicer” menu:
```
./src/gtk/dialog.cpp(199): assert "Assert failure" failed in EndModal(): either wxDialog:EndModal called twice or ShowModal wasn't called
```
```
ASSERT INFO:
./src/gtk/dialog.cpp(199): assert "Assert failure" failed in EndModal(): either wxDialog:EndModal called twice or ShowModal wasn't called

BACKTRACE:
[1] wxDialog::EndModal(int)
[2] Slic3r::GUI::AboutDialog::onCloseDialog(wxEvent&)
[3] wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&)
[4] wxEvtHandler::SearchDynamicEventTable(wxEvent&)
[5] wxEvtHandler::TryHereOnly(wxEvent&)
[6] wxEvtHandler::ProcessEventLocally(wxEvent&)
[7] wxEvtHandler::ProcessEvent(wxEvent&)
[8] wxWindowBase::TryAfter(wxEvent&)
[9] wxDialogBase::EmulateButtonClickIfPresent(int)
[10] wxDialogBase::OnCloseWindow(wxCloseEvent&)
[11] wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&)
[12] wxEventHashTable::HandleEvent(wxEvent&, wxEvtHandler*)
[13] wxEvtHandler::TryHereOnly(wxEvent&)
[14] wxEvtHandler::ProcessEventLocally(wxEvent&)
[15] wxEvtHandler::ProcessEvent(wxEvent&)
[16] wxEvtHandler::SafelyProcessEvent(wxEvent&)
[17] wxWindowBase::Close(bool)
[18] wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&)
[19] wxEvtHandler::SearchDynamicEventTable(wxEvent&)
[20] wxEvtHandler::TryHereOnly(wxEvent&)
[21] wxEvtHandler::ProcessEventLocally(wxEvent&)
[22] wxEvtHandler::ProcessEvent(wxEvent&)
[23] wxWindowBase::TryAfter(wxEvent&)
[24] wxEvtHandler::SafelyProcessEvent(wxEvent&)
[25] g_closure_invoke
[26] g_signal_emit_valist
[27] g_signal_emit
[28] g_closure_invoke
[29] g_signal_emit_valist
[30] g_signal_emit
[31] g_closure_invoke
[32] g_signal_emit_valist
[33] g_signal_emit
[34] gtk_propagate_event
[35] gtk_main_do_event
[36] g_main_context_dispatch
[37] g_main_loop_run
[38] gtk_main
[39] wxGUIEventLoop::DoRun()
[40] wxEventLoopBase::Run()
[41] wxDialog::ShowModal()
[42] Slic3r::GUI::about()
[43] wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&)
[44] wxEvtHandler::SearchDynamicEventTable(wxEvent&)
[45] wxEvtHandler::TryHereOnly(wxEvent&)
[46] wxEvtHandler::ProcessEventLocally(wxEvent&)
[47] wxEvtHandler::ProcessEvent(wxEvent&)
[48] wxEvtHandler::SafelyProcessEvent(wxEvent&)
[49] wxMenuBase::SendEvent(int, int)
[50] g_closure_invoke
[51] g_signal_emit_valist
[52] g_signal_emit
[53] gtk_widget_activate
[54] gtk_menu_shell_activate_item
[55] g_closure_invoke
[56] g_signal_emit_valist
[57] g_signal_emit
[58] gtk_propagate_event
[59] gtk_main_do_event
[60] g_main_context_dispatch
[61] g_main_loop_run
[62] gtk_main
[63] wxGUIEventLoop::DoRun()
[64] wxEventLoopBase::Run()
[65] wxAppConsoleBase::MainLoop()
[66] wxEntry(int&, wchar_t**)
[67] Slic3r::CLI::run(int, char**)
[68] main
[69] __libc_start_main
[70] _start
```

lukasmatena, 2019-05-16:
> there is another one, when going to the “About PrusaSlicer” menu and clicking on “Close”

What wxWidgets version you link against? This reminds me of https://github.com/prusa3d/PrusaSlicer/issues/1507.

ArchangeGabriel, 2019-05-16:
This is indeed 3.0.4 too, 3.1 is an unstable branch with a lot of breaking changes, so not publishable in distros for now. So I guess this is a duplicate of #1507 which should then be re-opened and this one closed.

bubnikv, 2019-05-16:
Do you get it for the other dialogs as well?

D:\src-perl\Slic3r-master\src\slic3r\GUI\AboutDialog.cpp:
  163  void AboutDialog::onCloseDialog(wxEvent &)
  164  {
  165:     this->EndModal(wxID_CLOSE);
  166      this->Close();
  167  }

D:\src-perl\Slic3r-master\src\slic3r\GUI\ConfigSnapshotDialog.cpp:
  159  {
  160      m_snapshot_to_activate = event.GetLinkInfo().GetHref();
  161:     this->EndModal(wxID_CLOSE);
  162      this->Close();
  163  }
  ...
  165  void ConfigSnapshotDialog::onCloseDialog(wxEvent &)
  166  {
  167:     this->EndModal(wxID_CLOSE);
  168      this->Close();
  169  }

D:\src-perl\Slic3r-master\src\slic3r\GUI\FirmwareDialog.cpp:
  883  			evt.Veto();
  884  		} else {
  885:             this->EndModal(wxID_CLOSE);
  886  			evt.Skip();
  887  		}

D:\src-perl\Slic3r-master\src\slic3r\GUI\KBShortcutsDialog.cpp:
  211  void KBShortcutsDialog::onCloseDialog(wxEvent &)
  212  {
  213:     this->EndModal(wxID_CLOSE);
  214      this->Close();
  215  }

D:\src-perl\Slic3r-master\src\slic3r\GUI\SysInfoDialog.cpp:
  172  void SysInfoDialog::onCloseDialog(wxEvent &)
  173  {
  174:     this->EndModal(wxID_CLOSE);
  175      this->Close();
  176  }

bubnikv, 2019-05-16:
Will it work if your remove this->Close() ?

ArchangeGabriel, 2019-05-16:
I indeed get this for the other dialogs as well. I’ll patch one of them and see.

ArchangeGabriel, 2019-05-16:
It does work after removing `this->Close()`.

bubnikv, 2019-05-16:
Thanks, I will get rid of the superfluous Close calls

bubnikv, 2019-05-16:
Hopefully fixed with https://github.com/prusa3d/PrusaSlicer/commit/57a7425b0fc096ea102c86da96aecc1ddcbda35b
Please reopen if you find out that it was not quite fixed.
Closing.

closed 2019-05-16 by bubnikv

ArchangeGabriel, 2019-05-17:
Can confirm this commit fix them all. Cheers!