prusa3d/PrusaSlicer #2254: Desktop icons refreshe on satup

jwvaughn, 2019-05-16:
### Version
2.0.0-rc+win64

### Operating system type + version
Win 10 Pro version 1809

### 3D printer brand / version + firmware version (if known)
Prusa MK3

### Behavior
When starting the slicer all the desktop icons are refreshed taking several seconds to do so.  When first run after starting windows, all the icons are changed to default icon then are slowly restored to what they should be. Subsequent starts causes all the icons to flash.

bubnikv, 2019-05-16:
this is very interesting and unusual. Would you please post a video? Thank
you.

jwvaughn, 2019-05-16:
I will work on that later today as I am about to head out now.

 

Jerry

jwvaughn, 2019-05-16:
Here you go - https://www.dropbox.com/s/n6qfmdr8pbgrtmd/Slic3rPE%20Glitch.mp4?dl=0

 

Jerry

Panayiotis-git, 2019-05-16:
Strange thing, but happens to me too! On Win7 64bit. And only with the PrusaSlicer 2.0.0-rc. Not with the Slic3rPE 1.42.0-beta2.

jwvaughn, 2019-05-16:
I agree – very strange and only with 2.0.0-rc.

bubnikv, 2019-05-17:
The flashing is happening due to the update of the desktop after the 3MF file is associated to PrusaSlicer. We shall only notify the desktop if the 3MF file was not associated yet. @enricoturri1966 will take care of it. Thanks for the report.

enricoturri1966, 2019-05-17:
This issue should be fixed by: https://github.com/prusa3d/PrusaSlicer/commit/a098f118b61b4276866efcaf3f8ea845af5b1ee8

FidelCapo, 2019-05-17:
This issue should be fixed in next release.
Closing.

closed 2019-05-17 by FidelCapo