prusa3d/PrusaSlicer #2255: [Feature request] more options in the modifiers (modifier of other parameters+sequential printing easiest profile optimization!)

otaolafranc, 2019-05-16:
### Version
Version 1.41.3+win64
### Operating system type + version
window 10
### 3D printer brand / version + firmware version (if known)
stock model MK3
_Is this a new feature request?_
yes
### Description of feature request
we are Prusa users we love bed meshes, don't we!?
changing between one brand and another of the same type of filament even if the two of them are high quality the profiles always need to be tunned for it, as different artifacts can be shown for same parameters for different brands.
there is a lot of different "issues" that are caused by more than one parameter, as the example that I give after, the stringing in a print could be caused by different things, retraction speed, length,temp and others....
if it would be possible to add some options of other parameters to the modifiers combining it with smalls sized prints and the complete individual objects option it could give us a highly easy and efficient way to optimize our printing profiles.
for example: 
retraction settings, what a lot of people test is changing the length and speed of the retraction, if a small object is designed to test the retractions settings of a material, we could do a mesh of objects like for example 3X3 wherein the columns the retraction speed (from 30-50mm/s) would increase each time and in the lines the retraction length(from 3-5mm), this will allow a combination of 9 different "couples" of the parameters. and would let the users a highly fine tune of the profiles.
(30mm/s,3mm) X(40mm/s,3mm)X(50mm/s,3mm)
(30mm/s,4mm) X(40mm/s,4mm)X(50mm/s,4mm)
(30mm/s,5mm) X(40mm/s,5mm)X(50mm/s,5mm)
after we arrive at a more or less optimal combination, we could do the combination of one of them with the temperature variation, and in two prints (more or less with small print times) we could perfectly erase the stringing.
this could be applied for different things, temperatures (temp also would allow a highly easy way to make a temp tower at the same time), speeds, retractions settings, fan speeds, extrusion multipliers, and others, this could help to create precise profiles for each brand in an easy reliable way seeing the influence of two parameters at the same time. for example for bridging, ghosting or other common problems.
it would be really easy to create .3MF files with the positioned copies and modifiers for each of them with only the user changing the values that he wants to study.
I hope I was clear enough and that this feature is added in the future :D thanks a lot!

nate98252, 2019-05-17:
Reinstate USB/serial interface

otaolafranc, 2019-05-17:
> Reinstate USB/serial interface
sorry? i didnt understand you.

nate98252, 2019-05-17:
The printing via USB feature has been removed. We are forced to use SD card only. So would like it put back in.
re·in·state
[ˌrēinˈstāt]
VERB
restore (someone or something) to their former position or condition.
"the union is fighting to reinstate the fired journalists"
synonyms:
restore · return to a former position · [more]

otaolafranc, 2019-05-17:
i understand perfectly the word "reinstate" what I didn't understand is why you were post it inside this feature request, you need to do a new post not an answer in a post not related with that.

bubnikv, 2019-08-01:
The current architecture does not support changing retraction or temp settings per object. It is doable, but quite a lot of work and complexity.

closed 2026-09-02 by otaolafranc