prusa3d/PrusaSlicer #2256: No tooltips for plater 3D view icons

jasontibbitts, 2019-05-16:
### Version
2.0.0-rc

### Operating system type + version
Linux, Fedora 29, self-compiled version of what will soon become the official Fedora packages.
Built with wxGTK3 3.0.4.

### Behavior
From reading a recent feature request I realized that there is supposed to be some sort of tooltip for at least some of the icons in the plater 3D view.  But I don't seem to have them.

I get tooltips for the various settings (under the Print Settings tab, for example, including the options, the locks and the bullets), just not on the plater icons.  And even on the plater screen, there are tooltips for things in the sidebar ("Simple", "Advanced", "Expert" and the cogs next to 'Print settings", "Filament" and "Printer", though not anything in the object list.  (Reading the source, I don't think there are supposed to be tooltips in the object list.)

But in the 3D view, there are no tooltips.  Just looking at Plater.cpp, I see the cube icon should have a tooltip of "3D editor view" and the icon next to it should have a tooltip of "Preview" (both with the shortcuts) but I don't see either.  Looking at GLCanvas3D.cpp, I can see the tooltips for the icons defined there ("Add...", "Delete", etc.) but they don't actually appear

mkoistinen, 2019-05-17:
👍

bubnikv, 2019-05-22:
The wxWidgets (the UI library) do not support all features quite well. 
@enricoturri1966 please look into it. Thanks.

jasontibbitts, 2020-01-17:
Just a note that I do have tooltips with 2.1.1 and with 2.2.0-alpha2, built with Fedora 31 (still using wxGTK3 3.0.4).  So I'll go ahead and close this.

closed 2020-01-17 by jasontibbitts