prusa3d/PrusaSlicer #2257: Buttons And Text Colors Description (Improved copy)

gryn, 2019-05-17:
### Version
1.42.0

### Operating system type + version
OS X / N/A

### 3D printer brand / version + firmware version (if known)
N/A

### Behavior
Copy in this dialog box is a bit obtuse and difficult to understand.
Suggested replacement copy:

> Slic3r allows the customization of many, many settings. Slic3r will periodically check for updates for your profiles -- these are called system profiles and have settings that will work for you 90% of the time.
> 
> You can also customize these settings! To help you maintain your customizations you may save these settings under custom profile with a different name. When the system profiles update, they will also update your customized profiles -- but only where you have not customized it.
> 
> When you are looking at your profile settings there are several icons between the label and the value of the setting. Below is a legend of what these mean:
> 
> [grey lock][white bullet] This setting is identical to the system profile.
> [black lock][back arrow] This setting is different from the system profile and to the current profile; pressing the lock returns to the system profile, pressing the back arrow returns to the current profile.
> [black lock][white dot] This setting is different from the system profile; pressing the lock returns to the system profile.
> [grey lock][back arrow] This setting is different from the current profile; pressing the back arrow returns to the current profile.
> 
> Additionally, these icons also appear near the top of the customization window. When pressed, they revert all settings to either the system profile [black lock] or the current profile [back arrow].

rtyr, 2021-01-28:
I believe this is no longer relevant. Closing.

closed 2021-01-28 by rtyr

gryn, 2021-01-28:
Thanks Rtyr, I agree the current dialog is much better than when I wrote this. I would still suggest the dialog contain some sort of copy introducing the basic concept of custom and system profiles, and how they interact over time. The current dialog is much better at describing what each icon means, but it presumes the user already understands the underlying concepts.