prusa3d/PrusaSlicer #2258: crash on slicing this content

akust, 2019-05-17:
### Version
_Version of PrusaSlicer used goes here_
PrusaSlicer 2.0.0rc
(crash reproduced on beta1 and beta2 too)

### Operating system type + version
macOS Sierra 10.12.6

_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_
[System Info.txt](https://github.com/prusa3d/PrusaSlicer/files/3189970/System.Info.txt)

### 3D printer brand / version + firmware version (if known)
i3mk3 

### Behavior
 _Describe the problem_
- it's a model with 4 colors (for palette2)
- crashs on slicing

_Steps needed to reproduce the problem_
- load attached things, call "Slice" or cmd+r

_Expected Results_
- well - sliced model, ready for export

_Actual Results_
- crash, "dead end"?

#### _note_

- If you have a guess what the reason is and/or know a workaround would be great if you could give me a tip in comment.

_Screenshots from __*PrusaSlicer*__ preview are preferred_
well.. real preview crashs but the pre-pre-view looks like this:
[image]

#### Project File (.3MF) where problem occurs
[c-base 3d fileB v16_full crashfiles.zip](https://github.com/prusa3d/PrusaSlicer/files/3189998/c-base.3d.fileB.v16_full.crashfiles.zip)

lukasmatena, 2019-05-17:
Thanks for the report, I reproduced the issue in current master. I believe the cause is the same as with #1266. Setting the support interface extruder explicitly to number 4 does not force the necessary toolchange on the wipe tower. Setting that to 0 makes the crash disappear (although that is not even a workaround).

I already attempted to fix this issue, but there was not yet time to test the fix properly, so it's not in the master yet. Your project seems to slice fine in my development branch - I'll attach an OSX build in a minute for you to try. Please, let us know if your model slices and prints fine when using that build. If so, I hope we will be able to push the fix into master after the PrusaSlicer2.0.0 final is out.

lukasmatena, 2019-05-17:
We gave it a bit more thought. The problem with your model seems to be the top of the tower, where it is so thin that some of the layers disappear during slicing. These empty layers then confuse the wipe tower and it fails to generate the necessary toolchange. So it appears to really be the same as #1266. 

You can try the development build I'm attaching. If you do, let us know if it works for you. But even if it slices, I doubt you'll be able to print the top of the tower, definitely not with the supports like they are now - they only support microscopic extrusions and are not needed there at all.
[PrusaSlicer-2.0.0-rc+36-lm-issue1266rebased-g7f1f03c-201905170813.zip](https://github.com/prusa3d/PrusaSlicer/files/3190856/PrusaSlicer-2.0.0-rc%2B36-lm-issue1266rebased-g7f1f03c-201905170813.zip)

akust, 2019-05-17:
thank you very much!
yes, after slicing with rc36 it didn't crash anymore (and the result also reminded me finally that the fragile long needle on top shouldn't be part of the print this way but e.g. aluminium part made with lathe tool).
I also can confirm that the model without the 'needle' on top slices correctly in 2.0.0-rc+
cool. thank you for the fast helpful response!

bubnikv, 2019-08-09:
@lukasmatena please look into the original project. It brings the PrusaSlicer 2.1.0-alpha1 down. Thanks.

bubnikv, 2019-08-09:
With the wipe tower disabled, PrusaSlicer 2.1.0-alpha1 slices this project fine. @lukasmatena is looking into this issue.

lukasmatena, 2019-09-10:
This should be fixed with https://github.com/prusa3d/PrusaSlicer/commit/2fef16c39aea81831ac376caae472b51082d6582 in the **dev** branch. The fix will be part of PrusaSlicer 2.2.0 release.

I'm closing the issue, the provided scene now seems to slice fine.

closed 2019-09-10 by lukasmatena

bubnikv, 2019-12-17:
Fixed with 2fef16c in PrusaSlicer 2.2.