prusa3d/PrusaSlicer #2260: [Feature request] New support type

drooboid, 2019-05-17:
### Version
Slic3r Prusa Edition
Version: 1.41.3+win64

### Operating system type + version
Win10 x64

### 3D printer brand / version + firmware version (if known)
Prusa i3 MK3S + FW 3.7.0

### New feature request
I have some models that I print on a daily basis and which are impossible to print without using support material. Right from the start I've sliced these models using Slic3r PE with support material enabled. Support layout can be seen on the preview:
[image]
Material I use is Prusament PETG and it turned out that sometimes it is really difficult to detach the print from the table because of this one-layer-thick zig-zag support layout.

Then I've tried to slice these models using original Slic3r, which has a different style of supports:
[image]
These prints were far more easily to detach from the table. Btw, they are printed faster and using less filament. Of course, they are not ideal, there are some issues with support material around model's 4 "legs". I prefer the way Slic3r PE deals with these issues, which has more control over support material options.

I'm wondering if it's possible to add this support material style into future versions of Slic3r PE so the user could choose between current style and the new one?

bubnikv, 2019-05-17:
Would you please describe more precisely what is the difference that is
important to you?

I see:

1) The support of the upstream slic3r are rounded.
2) There is a sheath around the supports generated by the upstream slic3r.
You can enable that in PrusaSlicer / Slic3rPE as well.
3) There are the "interface loops" generated by the upstream slic3r. You
can enable that in PrusaSlicer / Slic3rPE as well.

drooboid, 2019-05-17:
Thank you for the quick reply!

1. Supports have rounded corners (looks nice and smooth :) )
2. Supports take less space than supports generated by Slic3r PE (the prints can have more object's instances due to higher print density)
3. Supports don't have one-layer-thick regions which are hard to detach from the table. It makes stronger adhesion though (but it's unnecessary in my case).
4. Interface loops are the option in Slic3r PE, and I like it, because I cannot get rid of them using the upstream slic3r (and why I want to use Slic3r PE).

guestisp, 2019-05-17:
> Supports have rounded corners (looks nice and smooth :) )

supports are to be trashed away, why it's important for you to have nice and smooth supports?

drooboid, 2019-05-17:
> supports are to be trashed away, why it's important for you to have nice and smooth supports?

It's not as important as other points in my post above (2 and 3). I can live without rounded corners if the other things will be implemented.

bubnikv, 2019-05-17:
> Supports don't have one-layer-thick regions which are hard to detach from the table. It makes stronger adhesion though (but it's unnecessary in my case).

I don't understand. Where are single layer supports printed?

drooboid, 2019-05-17:
> I don't understand. Where are single layer supports printed?

Here:
[image]

bubnikv, 2019-05-17:
That is intentional to improve first layer adhesion.

drooboid, 2019-05-17:
> That is intentional to improve first layer adhesion.

I know that! :) But what if user doesn't need so much adhesion? 
Prints with supports generated by upstream slic3r are well balanced between adhesion and detachability.

guestisp, 2019-05-17:
The highlighted part is just 1 layer, which detachability issue do you have ? It's not connected to the printed part. Detachability from the place ? Then break the support, you don't have to remove it gently.....

bubnikv, 2019-05-17:
The supports generated by the upstream slic3r fuse to the top of the object, and sometime never finish to generate.

drooboid, 2019-05-17:
> The supports generated by the upstream slic3r fuse to the top of the object, and sometime never finish to generate.

What do you mean? I don't get it.

DL4APT, 2019-05-28:
I think the problem with the red marked area is if the print sticks relay well some part remain on the buildplate. And these remains are a real pain to remove as flexing the springsteel don't help and blunt force may damage the PEI surface. Maybe an option for 2 perimeters to connect them on the edge may help. Or an option to reduce the first layer extension by 25, 50 and 75%.

numanair, 2019-05-28:
I think those single layer areas marked in red work quite well with one exception. My build area is quite small and those areas tend to extend well past the model, often forcing me to rotate parts to fit.

bubnikv, 2021-09-23:
Supports were improved in PrusaSlicer 2.4.0-alpha1, see the change log.

closed 2021-09-23 by bubnikv