prusa3d/PrusaSlicer #2262: Slic3r-console and bed centering

luketaverne, 2019-05-17:
### Version
`version_1.42.0-beta2-16-g7f4e6328f` -> upstream
`version_1.42.0-beta2-21-gf8833cbdd` -> our fork [ubuntu-build branch](https://github.com/9t-labs/prusa-Slic3r). This issue refers to changes made in this branch.

### Operating system type + version
Ubuntu 18.04

### 3D printer brand / version + firmware version (if known)
n/a

### Behavior
We build prusa-slic3r without the gui interface, and are trying to implement bed centering in `slic3r.cpp`, similar to how mainstream slic3r works. We've implemented the framework for this in our branch [slic3r.cpp:381-386](https://github.com/9t-labs/prusa-Slic3r/blob/f8833cbddd780029fe232dbafe1b8868905dc268/src/slic3r.cpp#L381). 

Line 382-383 converts the `bed_shape` config to a `Slic3r::Polygon`, and I've printed and confirmed the values are correct at that point (`3.2e2` and `2.25e2` for an Ultimaker S5 bed). Line 384 should calculate the centroid of the polygon as a `Point`. I expect `1.6e2` and `1.125e2`, but receive values that are off by a factor of `1000000` (hence line 385, just needed to make it work). I've confirmed this by printing the values and by visualizing the resulting gcode in repetier host.

I've looked into the documentation for Eigen, as this seems to be some kind of internal type conversion problem, but I'm at a loss. Can anyone point me in the right direction for fixing this properly, and better understanding what causes this problem?

Many thanks in advance!

#### Project File (.3MF) where problem occurs
any file/ config, the centroid from the polygon is off by a factor of 1000000.

lukasmatena, 2019-05-17:
As for the cause of the problem, look at Polygon::new_scale(...) that is called at 382, that will lead you there quickly. Slicing core of PrusaSlicer heavily relies on Clipper library that uses fixed-point arithmetic to avoid floating-point precision pitfalls. PrusaSlicer custom types such as Point and Polygon therefore use integer coordinates internally and scaling is needed to achieve desired resolution.

closed 2019-05-17 by lukasmatena

luketaverne, 2019-05-17:
Thanks @lukasmatena! I'll look into it.

If anyone is interested in the bed centering, it will continue to exist in our `ubuntu-build` branch. Let me know if there is interest in a pull request at some point.

luketaverne, 2019-05-31:
For anyone referencing this in the future, there is an `unscale()` function, that you can use like this:

```cpp
auto unscaleCentroid = Point(unscale<double>(centroid.x()), unscale<double>(centroid.y()));
```

Thanks again @lukasmatena!