prusa3d/PrusaSlicer #2263: [feature request] Faster g-code export

Hoeze, 2019-05-17:
Hi, the current g-code generation is single-threaded and runs quite slow.
Would it be possible to parallelize this task?
On my 6core/12threads cpu this task would optimally reduce from three hours to 30min.

### Version
2.0.0 RC

### Operating system type + version
Arch linux

SckyzO, 2021-04-09:
Hello, 

Any date for this feature ? 

it is not possible to wait so long to generate a gcode ... S3D or cura generate my gcode in 10 secondes, et prusa 5 minutes I use a 20 cores / 40 threads for working ... 😄 

Thank you very much

bubnikv, 2021-09-25:
G-code export was parallelized to some extent (usually to three threads) and optimized. On my 12 core desktop the slicing often runs around 3x-4x faster.

closed 2021-09-25 by bubnikv

wuast94, 2022-12-05:
so would be nice to use n threads instead of (hardcoded?) 3

wuast94, 2022-12-05:
> you guys did not provide data to test on. Your particular slow down may be specific to some feature, for example I have just significantly optimized G-code export for "only_retract_when_crossing_perimeters" enabled.

when the gcode generator or parts of it only runs on 3 cores i dont think its neccesary to provide data, its a overall topic and would be overall nice if the slicer uses all system resources available instead just some of it.