prusa3d/PrusaSlicer #2267: There is an error in the menu during the Korean translation process.

ulsanether, 2019-05-18:
I am making a translation file. 
[image]

However, there is a problem with GUI during translation.
I would like to solve this. Please help.
[image]
[image]

bubnikv, 2019-05-19:
I believe I have found a bug in the font initialization for Korean
9fca67d2812090befec5a159facc16b22c1d5b85

Still this bugfix did not help. Would you please test your texts with the font
resources\fonts\NotoSans-Regular.ttf
?

If it works, then we likely need to extend the function
ImGuiWrapper::set_language(const std::string &language)
to select correct glyph ranges for rendering in the 3D scene.

ulsanether, 2019-05-20:
Translation file attached. I'd like an inspection.

[prusaslicer_ko_KR.zip](https://github.com/prusa3d/PrusaSlicer/files/3199304/prusaslicer_ko_KR.zip)

bubnikv, 2019-05-20:
>  Translation file attached. I'd like an inspection.

Sorry, I don't understand what you mean.

You may just replace your files in the resources/localization/ directory
and the application shall accept them.

closed 2019-05-21 by ulsanether

bubnikv, 2019-05-21:
You closed this issue. Has the issue been solved? What was the problem and what was the solution?