prusa3d/PrusaSlicer #2268: Erratic behavior of scrolling sliders

liftbag, 2019-05-18:
### Version
_Slic3rPE-1.42.0-beta2+ and PrusaSlicer-2.0.0-rc+_

### Operating system type + version
_macOS 10.14.4 (18E226)_

### 3D printer brand / version + firmware version (if known)
_not relevant_

### Behavior
* All the scrolling sliders exhibit irregular behavior. In most cases they cannot be dragged with the mouse pointer, the only way to scroll is with the mouse wheel. 
* _Steps needed to reproduce the problem_
  * _pull down one of the presets list in the Print Settings/Filament settings/Printer settings and try do drag the scrolling slider with the mouse pointer_
* _Expected Results_
  * _the preset list should be scrolled_
* _Actual Results_
  * _In most cases the list remains immobile. Sometimes it starts functioning regularly and then returns immovable at the next activation_

I suppose it's a macOS issue only, if not even limited to my personal installation.
Look at the screen record below.

https://youtu.be/PVNA641MGDw

JohnOCFII, 2019-05-18:
Can confirm.  Using 2018 MacBook Pro with Mojave 10.14.4.  You can scroll using two fingers on the TrackPad, but trying to grab the scroll indicator in the scroll bar on the right side of the window has no effect.  Checking now, this is also an issue with Beta 2 as well as the last released version 1.41.2+.

bubnikv, 2019-05-19:
I have a feeling that the issue was always there. What strikes me is that nobody noticed until now. Once @liftbag found out, everybody will notice :-)

It would help, if you guys could find the first version where it broke.

JohnOCFII, 2019-05-19:
I tested back to 1.35.5.  The UI is different there, and NO scrolling works.  It fails on 1.38.3 the same way it fails now, so behavior has been the same since then (December, 2017).

liftbag, 2019-05-19:
> I have a feeling that the issue was always there. What strikes me is that nobody noticed until now. Once @liftbag found out, everybody will notice :-)
> 
> It would help, if you guys could find the first version where it broke.

I'm sorry, but I got on board late with Slic3r PE.

bubnikv, 2019-05-20:
>  I tested back to 1.35.5. The UI is different there, and NO scrolling
works. It fails on 1.38.3 the same way it fails now, so behavior has been
the same since then (December, 2017).

Thanks for helping out. I suppose it is not a pressing issue then.

JohnOCFII, 2019-05-20:
I'm not the OP, but I'd agree, it is just something to watch for with newer UI widgets.  Not pressing.

bubnikv, 2019-05-20:
We plan to send one printer to the man, who implemented the OSX wxWidgets
port, let's hope we will make him interested :-)

closed 2026-09-29 by prusaslicer-bot[bot]