prusa3d/PrusaSlicer #2269: “PrusaSlicer.app” can’t be opened because it is from an unidentified developer.

cassianoleal, 2019-05-18:
### Version
2.0.0-rc

### Operating system type + version
macOS (any version)

### 3D printer brand / version + firmware version (if known)
Irrelevant

### Behavior
* When opening PrusaSlicer, macOS tells me that `“PrusaSlicer.app” can’t be opened because it is from an unidentified developer.`.

* _Steps needed to reproduce the problem_
  * Open `.dmg` file, drag app to `Applications`, open it.

* _Expected Results_
* PrusaSlicer opens

_Is this a new feature request?_
No, this is a security bug.

There's good reason for the signing process and I like to respect that. Apple Developer Program costs €99 (inc. VAT) per *year*. 

Signing gives users a much better experience and sense of security since then we can trust that the file that's just been downloaded has not been tampered with by a third party.

JohnOCFII, 2019-05-18:
This is just a release candidate.  The last official release of Slic3r PE was signed, and I would expect PrusaSlicer will be signed once the Release Candidate process is complete.

bubnikv, 2019-05-19:
All our OSX builds shall be signed. There was a typo in an automatic build script. Thanks for notifying us.

This is the response I am getting from the codesign command line tool after the build script fix.

$ codesign -dv --verbose=4 "PrusaSlicer/PrusaSlicer.app/"

Executable=/Volumes/PrusaSlicer/PrusaSlicer.app/Contents/MacOS/PrusaSlicer
Identifier=com.prusa3d.slic3r/
Format=bundle with Mach-O thin (x86_64)
CodeDirectory v=20200 size=108863 flags=0x0(none) hashes=5436+3 location=embedded
Hash type=sha1 size=20
CDHash=68a97b1bb548b233f40b197e542bbf9f8ad1fd11
Signature size=8542
Authority=Developer ID Application: Prusa Research s.r.o. (4LVTA6C5N9)
Authority=Developer ID Certification Authority
Authority=Apple Root CA
Timestamp=18 May 2019 14:19:09
Info.plist entries=14
TeamIdentifier=4LVTA6C5N9
Sealed Resources version=2 rules=12 files=209
Internal requirements count=2 size=1280

closed 2019-05-19 by bubnikv