prusa3d/PrusaSlicer #2273: Rotation gizmo for support enforcer/blocker are not parallel to XYZ

eowindel, 2019-05-18:
### Version
2.0.0 RC

Hello,

There is a problem with rotation gizmo for support enforcer / blocker after a stl rotation.

Procedure to reproduce
Load a STL file -> rotate the file on X or Y -> Add a box support enforcer -> rotate the box. The gizmo to rotate the box are still parallel to STL original import plan, it shall be parallel to X Y Z.

[example.zip](https://github.com/prusa3d/PrusaSlicer/files/3194253/example.zip)

[image]

ddm-j, 2022-08-15:
Adding my input. Would be nice to have "Global/Local" coordinate system selection for modifiers. I think this would solve the problem.

Or allowing to select transformation coordinate system according to a modifier instance, rather than the parent object.

stepikovo, 2022-08-24:
We are aware of the issue, it's on the to-do list, which is, however, very long.

closed 2026-09-29 by prusaslicer-bot[bot]