prusa3d/PrusaSlicer #2274: Infill modifier splits top layers

rennne, 2019-05-18:
### Version
2.0.0. RC

### Operating system type + version
Windows 10

Added a modifier box to make a denser infill.
But it changes the top layers like ist can be seen in the picture.

[image]

it seems the split on the top layer is located where the box ends:

[image]

liftbag, 2019-05-20:
This doesn't happen if you reduce the modifier box Z height below the top layer.
But I agree that it would be convenient not to have to do it.

rennne, 2019-05-20:
Yes I know.
Like with this part where mechanical stability is important, it would be essential to have really all top and bottom layers without a split.

It is very difficult to get this box so that it does hit all of the infill and not a few of the inner top or bottom layers, and would only work like in this case where the surface is not tilled or rounded.

The only proper workaround I would see right now, would be to model the infill-zone within the cad software, and load it as modifier shape.

liftbag, 2019-05-20:
> The only proper workaround I would see right now, would be to model the infill-zone within the cad software, and load it as modifier shape.

This is what I do in most cases.

lukasmatena, 2020-02-11:
I believe this was a duplicate of #2982 and as such should be fixed in PrusaSlicer 2.2.0-alpha4. You can retest.
Closing the issue. If anyone disagrees, it can be reopened.

closed 2020-02-11 by lukasmatena