prusa3d/PrusaSlicer #2275: Gap in model with scale factor 200%

cappuccinoFan, 2019-05-18:
### Version: 
2.0.0-rc+linux64
Build: PrusaSlicer-2.0.0-rc+linux64-201905141708
Operatin system Unix
System Architecture 64bit
System Versuib Kinux 4.15.0-50-generic
OpenGL installation
GL version: 4.6.0 NVIDIA 390.116
Vendor NVIDIA corporation
Renderer Geforce GTX 1060 6GB/PCIw/SSE2
GLSL version: 4.6.0 NVIDIA

### Behavior

Problem:
The sliced model (double size) does have a gap (free space inside sliced model) and makes it unprintable.

I wanted to do:
I wanted to print out the Buga (BundesGartenSchau-Zwerg) in a bigger size.
The model can be downloaded at:
https://www.buga2019.de/we-dokumente/zwerg-3d-druckdatei/Buga-Zwerg-Bundesgartenschau-Heilbronn-2019.stl?m=1550501694&

Normal printing works fine.

Problem:
Problem appears when size is increased to 200%  (scale factor same for x,y,z  = 200%) 
Afterwards slice again and the gap inside the face appears. (see screenshot)

[image]
e the face)

* Same result for different slicing parameter (0,15mm detail, 0,2mm fast, ...
Expected result:
No gap in sliced model
* _Actual Results_
Gap in sliced model
 see screenshot

_Is this a new feature request?_   : 
no, bug-report

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
[buga.zip](https://github.com/prusa3d/PrusaSlicer/files/3194313/buga.zip)

BerndJM, 2019-05-18:
Das Buga-STL ist von Haus aus kaputt - habe das in der selben Situation (hochskalieren) feststellen müssen. Hab's leider erst beim Drucken gemerkt ;-)).
Dafür kann der Slicer beim besten Willen nix. Wenn im Original eine minimale Lücke ist, dann ist die halt auch 200% größer und macht sich bemerkbar. Aus 0.05mm wird dann halt gleichmal 0.1mm
Ich hänge dir mal eine reparierte Version an.

The Buga STL is faulty - experienced that myself in the same situation (upscaling) as you.
You can't blame the slicer for this - if there is a gap, let's say 0.05mm in the original file than this gap became 0.1mm at 200% scaling. Attached a repaired version.

[Buga-Zwerg-Bundesgartenschau-Heilbronn-2019 (repaired).stl.zip](https://github.com/prusa3d/PrusaSlicer/files/3194350/Buga-Zwerg-Bundesgartenschau-Heilbronn-2019.repaired.stl.zip)

rtyr, 2021-01-28:
Broken model. It is ok after model repair. Closing.

closed 2021-01-28 by rtyr