prusa3d/PrusaSlicer #2276: Layer height modifier not respected

WarrenSchultz, 2019-05-18:
### Version
2.0.0-rc+win64-201905141708

### Operating system type + version
Win10 1809

### 3D printer brand / version + firmware version (if known)
Mk3S

### Behavior
Adding a cylinder modifier to an object with threading and then adding the Layers and Perimeters fields to the modifier, Bottom, Detect Bridging, Perimeters, and Top will all trigger a re-slice and have the proper effects. Adjusting the Layer Height has no effect on the sliced model, nor does it trigger the need for a re-slice.

Master3See, 2019-05-19:
Hi, this has been addressed before. Setting layer height via a modifier isn't supported by PrusaSlicer intentionally. I believe that @bubnikv mentioned they were going to remove the option from the modifier menu in a future version.

The reason given was because layer heights can only change on an entire layer.

There is another built in way to modify layer heights in PrusaSlicer, you access it by selecting the object you wish to modify, and choosing the right most option on the top bar in the plater.

WarrenSchultz, 2019-08-09:
Layer height table takes care of this. Thanks!

closed 2019-08-09 by WarrenSchultz