prusa3d/PrusaSlicer #2278: support_material_contact_distance: numeric input not valid

guestisp, 2019-05-19:
Previous issue was closed even without reading it, it was considered a suggestion but it's a bug triggering an error due to a wrong default value, so, as it is still closed, i open this new one hoping in more attention

https://github.com/prusa3d/PrusaSlicer/issues/2261

bubnikv, 2019-05-19:
There are 668 issues open, which are a lot more imporant than this. The "issue" you are reporting was never ever any problem to anybody for 8 years of the existence of the Slic3r project. I believe you are just too picky.

Closing.

closed 2019-05-19 by bubnikv

guestisp, 2019-05-19:
But is still a bug. Closing bug issues because you don't think they are important is kinda stupid

guestisp, 2019-05-19:
And no, is not true saying that this issue hasn't annoied anyone in 8 years. This is a regression, in 1.41.3 is working properly, in 1.42.0-beta1 is working properly, in 1.42.0-beta2 is working properly, in RC1 is broken.

so, please do your homework, check this issue and keep it open, maybe some volunteer with spare time available is willing to fix, but if you close issues based on your mood, nobody would be notified.

Keep in mind that this is an open source community-driven project, not your personal home project, you can't close issues based on your mood. 

A bug is a bug, the relative issue must stay open. If you are not interested, then don't fix it, but don't close issues hiding them from volunteers.

lukasmatena, 2019-05-21:
@guestisp 
We managed to reproduce the bug eventually (we did not understand your problem correctly at first, hence the fast close). The bug was related to localization and it appeared on your system because you are (presumably) running in italian. The bug should be fixed by the commit mentioned above. Thanks for the report and we're glad it's settled.

While we appreciate users looking for bugs so we can fix them, the cooperation could be even better if the reports were more verbose and easily reproducible, i.e. they would contain a 3mf if it could be project related issue, ideally a screenshot or at least very good description of everything related and always a platform info and PrusaSlicer version. Please understand that there are loads of issues open and we don't have time to guess what the user meant and whether it is actually a bug (quite often it isn't). The misunderstanding above was avoidable had the issue report contain more info. Don't take this as critics of yourself, this is a general statement.

> Keep in mind that this is an open source community-driven project, not your personal home project, you can't close issues based on your mood.

Actually, this is Prusa company repository, @bubnikv is the project lead and unlike the "community", he is responsible for it. Please, don't boss him around. I can assure you the issue wouldn't be closed if we comprehended it correctly the first time. Open-source is about sharing source code and allowing anyone to base their work on ours. I believe we honor these principles well enough.

guestisp, 2019-05-21:
At first, we are greatly thankful for your awesome work. We think PrusaSlicer it's the best slicing software available, it's very fast, very easy to understand and so on. There is only a huge missing: an animated preview, but it doesn't matter, it's a free software, so . Everything else is almost perfect and it's thanks to you, this is for sure.

Said that, we know that this is a Prusa company repository, thus you are not a simple "volunteer" contributing to a thirty-party software. This is your fork, and, as Prusa employees, you are being paid to develop and handle it (now more than in the past, as you placed the Prusa word in the software name, renaming if from "Slic3r Prusa Edition" to "PrusaSlicer")

We are volunteers, noone is paying us to use/handle/manage PrusaSlicer. We do this on our own in our spare time. Personally, i'm being paid for a totally different work. What I do is during my personal spare time or during my working hours if I have a little bit spare time available between my paid jobs

Saying, in almost every opened issue, that we didn't do our homeworks, is (at least for non native english speaker, but I think it's the same also for native speakers) rude, offensive or similiar. As wrote, we are not paid to do anything, it's just a contribution done in the spare time. Who is supposed to do the homework, it's you, as paid Prusa employee working on a Prusa software.

Obviously, you are free to ask more details to the contributor, but you can't **force** them to do the work for you saying "do your homework", at least if you know the meaning of "do your homework". 
Are you my work leader to boss me a job ? I don't think so, I don't see the word Prusa in my monthly paycheck

An open source software only survive based on community contributions. A contribution is a cotribution, not a paid job. You can't force volunteers to do the work if they were you. Yes, a detailed contribution is more useful than a short one, but short contributions are still much much much more useful than no contributions, in an open source software.

Most of the time seems that issues are opened just to hurt you or as aggression. It's not like this. Issues are often opened as "reminder" to say: "hey, there is a bug here, it has to be fixed **sooner or later**, just don't forget it". no one is saying "fix this right now" or "fix this immediatly"
Often, the issue is opened as reminder for the contributor itself "i open this quickly right now, then when I have more time available, i'll add details, but at least I don't forget it"
Often (and this is my case for projects where I'm actively partecipating), the issue is opened just as reference for an upcoming pull-request "i open this quickly, no need for a detailed explaination as the fixing pull-request is coming because is something i've already fixed. There is no need to details you what to fix, i've already done that."

If the issue is not clear enough, just ask for more info, don't close them without reason and even without asking for more details.

I think you (at least one particular developer) should be more polite with the community and with the contributors.

guestisp, 2019-05-21:
Anyway:

> The bug was related to localization and it appeared on your system because you are (presumably) running in italian. 

Yes, it was related to the Italian, but it wasn't present in previous releases

> The bug should be fixed by the commit mentioned above.

Yes, fixed, I can confirm