prusa3d/PrusaSlicer #2279: Crash when setting 100% infil

Kolous, 2019-05-19:
### Version
2.0.0-rc

### Operating system type + version
Win 7 64

### 3D printer brand / version + firmware version (if known)
MK2.5s (3.7.1) MMU2s (1.0.5)

### Behavior
When setting 100% infil in the attached scene program crashes with following error: 

9:20:28: Internal error: boost::bad_format_string: format-string is ill-formed

[image]

#### Project File (.3MF) where problem occurs
[Body1.zip](https://github.com/prusa3d/PrusaSlicer/files/3194856/Body1.zip)

liftbag, 2019-05-19:
I think it's because 100% infill with cubic infill pattern is not allowed. In macOS PrusaSlicer don't crashes and a friendly warning asks to be able to change the type of infill.

[image]

Kolous, 2019-05-19:
Last version asked to change infill pattern. Now it crashes.

bubnikv, 2019-05-20:
Does it happen if you switch your language to English?

bubnikv, 2019-05-20:
It is an issue in translations, where the percent mark in templates has to
be doubled, and one of the percents got lost. We will fix that in the final
release. Thanks for the report.

bubnikv, 2019-05-20:
Fixed in the 2.0.0 final

closed 2019-05-20 by bubnikv