prusa3d/PrusaSlicer #2280: unable to build on linux due to missing PrecompiledHeader

davidelang, 2019-05-19:
### Version
_Version of PrusaSlicer used goes here_
git 9fca67d2812090befec5a159facc16b22c1d5b85

### Operating system type + version
Ubuntu 19.04

### Behavior
:~/git/PrusaSlicer/src/build$ cmake ..  -Wno-dev
-- qhull Version: 7.2.0 (static linking)
-- Parallelization with OpenMP
-- Boost version: 1.67.0
-- Found NLopt in '/usr/lib/x86_64-linux-gnu'.
-- Using NLopt include directory '/usr/include'.
-- Using NLopt library '/usr/lib/x86_64-linux-gnu/libnlopt.so'.
CMake Error at libslic3r/CMakeLists.txt:4 (include):
  include could not find load file:

    PrecompiledHeader

CMake Error at CMakeLists.txt:201 (install):
  install TARGETS given no RUNTIME DESTINATION for executable target
  "PrusaSlicer".

-- Configuring incomplete, errors occurred!
See also "/home/dlang/git/PrusaSlicer/src/build/CMakeFiles/CMakeOutput.log".
See also "/home/dlang/git/PrusaSlicer/src/build/CMakeFiles/CMakeError.log".

bubnikv, 2019-05-20:
I suppose we shall make the precompiled headers on Linux optional. Some gcc
versions are not stable with precompiled headers anyway.

jasontibbitts, 2019-05-20:
As far as I know they are already optional.  In Fedora I'm building with `-DSLIC3R_PCH=0`.

vojtechkral, 2019-05-20:
The error is that the CMake module file for PCH has not been found, apparently, which is really weird, since we keep it in the `cmake/modules` directory. I wonder if OP invoked CMake in some nonstandard way or did some modifications that might've caused this...

bubnikv, 2021-09-25:
We don't have resources to support linux self-compilation.
We have updated our guide.
https://github.com/prusa3d/PrusaSlicer/blob/master/doc/How%20to%20build%20-%20Linux%20et%20al.md

closed 2021-09-25 by bubnikv