prusa3d/PrusaSlicer #2283: [feature request] gyroid supports

otaolafranc, 2019-05-20:
### Version
PrusaSlicer 2.0.0-rc2

### Operating system type + version
windows 10
### 3D printer brand / version + firmware version (if known)
MK3

#### feature request
as the gyroid infill a support pattern gyroid would be cool, the infill even with really low % presents is reliable and also the smooth printing of it makes more reliable the printing in general.

wavexx, 2019-06-19:
Not to detract from this suggestion, but "corrugated" supports would probably be a better middle-ground between speed and reliability (gyroid is faster than hexagonal, but definitely not as a fast as a plain sinus).

rtyr, 2021-01-29:
I believe it would be very hard to remove.

foreachthing, 2021-01-29:
> I believe it would be very hard to remove.

Why? Because it's strong? 
That actually might help, I think.

NoOneAtHome, 2021-09-01:
I too would like to see gyroid support.  I am printing the [head(amame) headphones](https://shop.vectorfinesse.com/).  It uses gyroid supports that you don't remove as an acoustic feature of the headphone cones and I am not a fan of Cura.

gopgnz, 2021-09-18:
> 
> 
> I too would like to see gyroid support. I am printing the [head(amame) headphones](https://shop.vectorfinesse.com/). It uses gyroid supports that you don't remove as an acoustic feature of the headphone cones and I am not a fan of Cura.

I would second that as someone who just received his Head(amame) kit as well. :) I love PrusaSlicer and it's my slicer of choice now.

buzkill420, 2021-09-19:
Bump. Was hoping someone would know how  to add this feature, as i dont mind compiling my own version. Seeing as there is gyroid infill, seems it might be as simple as adding a few lines of code somewhere.

rtyr, 2021-09-19:
> Why? Because it's strong?
> That actually might help, I think.

We did some experiments with adaptive cubic supports. Removing such supports was unfortunately not pleasant experience.

gopgnz, 2021-09-19:
So, while gyroid supports are not (yet) a thing in PrusaSlicer I thought I should take advantage of gyroid infill.
See the result I got (including notes on settings) here: https://imgur.com/a/Cj1roZc
And here's my post on this in the head(amame) sub-reddit: https://www.reddit.com/r/VectorFinesse/comments/pqrok7/prusaslicer_gyroid_supports_workaround/

n8bot, 2021-11-26:
What does a guy have to do around here to get some gyroid supports, amirite?

closed 2026-09-02 by otaolafranc