prusa3d/PrusaSlicer #2285: "Generating G-code" in the status bar, App hung

Looxoor, 2019-05-20:
### Version
2.0.0 RC2

### Operating system type + version
MacOS 10.14.5

### Behavior
* This file takes multiple hours to generate a G-code for, during which time the UI gets unresponsive until it suddenly springs back to life.
* I took this to mean that I could finally press the "Export G-code" button, only to be met with an error message "Another export job is already running".
* I'd like a working status indicator (it's at 95% and been there since it began generating g-code automatically) and that the export button is inactive when it's useless to press it.
* Ideally though faster g-code generation. It sliced in mere seconds and I've never with earlier versions experienced such slow behavior.

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_

[Stand.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/3198751/Stand.3mf.zip)

Jebtrix, 2019-05-21:
Win7 Pro 64bit
"Avoid crossing perimeters" option is the cause.  Unchecking that it will slice in seconds. I can confirm it does get stuck at 95% for at least over an hour with it enabled.

Looxoor, 2019-05-22:
Doh!

Of course, sorry, closing!

closed 2019-05-22 by Looxoor