prusa3d/PrusaSlicer #2289: Feature type on verbose comments

cfelicio, 2019-05-20:
### Version
2.0.0

### Operating system type + version
Windows 10 64 bit

### 3D printer brand / version + firmware version (if known)
Prusa MK3

### Behavior

Right now, when verbose is enabled, I only see on the gcode 2 types of features:

G1 X142.589 Y115.413 E4.21475 ; perimeter

or

G1 X144.666 Y116.038 E4.37289 ; infill

Is it possible to get more specific details, like we get from the GUI?

[image]

bubnikv, 2019-05-24:
Mosaic guys are asking for it to support their purge into infill.

rtyr, 2021-01-29:
Should be covered by new comments, implemented for g-code viewer in 2.3.0. Closing.

[image]

closed 2021-01-29 by rtyr