prusa3d/PrusaSlicer #2291: [Feature Request] Control loaded object visibility and/or print-inclusion

clintonefisher, 2019-05-20:
In many cases, it's handy to be able to temporarily make an object invisible and/or prevent it from being sliced.  Simplify 3D does this for both printing and for object editing visibility, however, I think PrusaSlicer could do it even better. 
The user-interface method I would use would be two checkbox columns in the object list on the right for each object:   "Visible?" and "Print?"
This would be very handy for many situations, such as projects that have several variants one might want to switch between, or maybe to show/hide test/calibration objects associated with the project.

bubnikv, 2019-05-21:
We handle this by allowing one to pull the object fully outside of the print bed. Such models are ignored by the slicing engine. Does it work for you?

PRGeno, 2019-06-13:
This would be an extremely useful feature and would eliminate another important differentiating feature that S3D has over Prusa Slicer.  

The need cannot be satisfied by dragging objects outside of the print bed which defeats the major use case.  After a bed has been carefully laid out, sometimes only some of the objects need to be printed in the current iteration.  It would be unproductive to drag models off the bed slice, output, and then place them back on the bed precisely again.   Likewise saving the project, removing what isn't needed in this iteration, and then reloading the project again.

Also there are many instances where temporarily hiding objects on the bed that are distracting, or in the way, while working on other objects would be very useful.

This is easily done in S3D and used maybe more often than the S3D Undo feature.  I strongly second the motion to add this important feature.

bubnikv, 2019-08-01:
Undo is being released with the PrusaSlicer 2.1.0-alpha1.

@YuSanka and @enricoturri1966 are working on suppressing selected objects from the print. The feature will likely be released with PrusaSlicer 2.2.

bubnikv, 2019-08-08:
Implemented by the upcoming PrusaSlicer 2.1.0-alpha1. Closing.

closed 2019-08-08 by bubnikv