prusa3d/PrusaSlicer #2293: [Feature Request] Infill (external and internal) angle lists

clintonefisher, 2019-05-20:
Simplify 3D has a handy feature, for both internal infill and external top & bottom infills, that allows the user to specify a repeating list of orientation angles to cycle through for each layer.

This is really handy let's say if one wants top & bottom layers to align with hexagonal infill.  
One can choose a repeating list of 0, 120, and 240 degrees for the top and/or bottom infills. 

As it stands now, PrusaSlicer only supports one user-set angle for both internal and external infill, and then automatically selects the alternating angle at 90 degrees from the specified angle.

RcBischof, 2020-01-25:
Since this hasn't been given a response in almost a year, I just wanted to second this request. Both Simplify3D and Cura support a list of infill angles that the slicer cycles through. Unfortunately for my use case, I need the PrusaSlicer support of the MMU2S but also need this variable infill direction capability. 

I have spent a good deal of time looking at the generated g-code, custom functions in both slicers, and the individual source code for both Cura and PrusaSlicer trying to figure out how to merge these features. However I'm not a programmer and only have a basic knowledge of C++.

I hope this is something that can be implemented soon so I don't have to resort to manually creating height zones for each direction.

bubnikv, 2020-02-09:
>  but also need this variable infill direction capability.

what for?

RcBischof, 2020-02-09:
I am trying to simulate natural Bouligand structures found in arthropod exoskeletons by continuously rotating layer direction.

This is possible in Cura by simply specifying the infill direction as a list [0,1,2,...,359,360].

e-SonOfAnder, 2023-01-21:
Will admit to not expecting to see this sitting around as a four year old feature request with no responses or movement.

One of the many reasons to desire setting the top/bottom line directions (as it is called in Cura) is to allow a better top surface of a print without using ironing, which adds time and material cost, and can often be avoided this way. For just one example showing the results of this, you can reference the following YouTube video: https://www.youtube.com/watch?v=jkAqTWmVRVA

For a TL;DR image comparison: https://imgur.com/a/6TdPhe3

JPK-MG, 2023-09-18:
I agree with implementing granular control of the top/Bottom (horizontal shell) line directions. 

Printing the top 2 layers at the same angle prevents visible crosshatching on the topmost layer - this goes especially for shiny filaments, like Silks, or Metallics. If the top 2 layers get the same line directions, the top surface can be tuned to a nice mirror(ish) surface.

JPK-MG, 2023-09-18:
I agree with implementing granular control of the top/Bottom (horizontal shell) line directions.

Printing the top 2 layers at the same angle prevents visible crosshatching on the topmost layer - this goes especially for shiny filaments, like Silks, or Metallics. If the top 2-3 layers get the same line directions, the top surface can be tuned to a nice mirror(ish) surface.

Cura has the option to set the angle for each layer (even though the ordering system is buggy and takes a bit of figuring), but the most valuable part of Cura's options is setting Top Surface Skin Layers, which allows you to set the number of topmost "skin" layers - in my case, 2 - then has a full set of options for those 2 layers, including line directions - in my case, 0°.

Hat's off to anyone that figures out and implements top "skin" options, to control those line directions!

Braintoe, 2025-09-27:
Let me bump this one up. 

My reason: depending on the model it is beneficial to be able to define the direction of the infill in order to optimise the model for its task. In Prusaslicer however, this automatically means you have to live with the top or bottom layer chaninge direction as well since the same angle setting controls both features. Thus, you have to live with one or another comprimise.

Braintoe, 2026-09-06:
Still interested in that (see last comment) therefore adding this comment on order to avoid closing the issue.

Jan-Soustruznik, 2026-10-01:
Hello, thank you for update, According to new [policy ](https://github.com/prusa3d/PrusaSlicer?tab=contributing-ov-file#issue-tracker-policy)I closing this feature request and converting it into a Discussion thread to hear what the community thinks about this feature. Thank you, and have a nice day.