prusa3d/PrusaSlicer #2294: [Feature Request] Add ability to create lambda objects (box, cylinder, etc.) outside of imported objects

clintonefisher, 2019-05-20:
It would be quite handy for quick test and calibration prints to be able to create a lambda object, much like we do for modifiers, etc, and have it be a primary object for printing - without having to import an external object and then add to it.    Seems most of the code is already there to be able to add this feature.

wavexx, 2019-06-13:
I concur. I have some simple stl files with a cylinder, box, sphere that I keep around for several reasons, but mostly for running a test print. In several cases though I use them as sacrificial objects to avoid overheating small layers.

Kind of silly that we need to load a box. It would be very useful to create simple solids the same way we can add modifiers.

wavexx, 2019-12-23:
I noticed you can actually do this now, right-clicking on the build plate -> Add Shape -> [box, cylinder, sphere]. I don't know when this got introduced, but that definitely satisfies my need!

lukasmatena, 2019-12-23:
This was implemented by @YuSanka for the 2.2.0 release (see also #2865). Closing this one too.

closed 2019-12-23 by lukasmatena