prusa3d/PrusaSlicer #2296: Add TEXT part

ZeGuigui, 2019-05-20:
### Version
PrusaSlicer 2.0.0+

### Operating system type + version
MacOS Mojave 10.14.1

### 3D printer brand / version + firmware version (if known)
MK3S + MMU2S

### Behavior
This is a Feature request

It would be nice to be able to add a text as a part to an existing object the way we can already add a box, cylinder, sphere or slab. We should be able to choose font face, font weight, size, extrusion height.

At the present time I use a simple OpenScad script to generate a STL that I add to the project.

#### Project File (.3MF) where problem occurs
No project file. This is a feature request!

Prusa-Support, 2022-05-17:
From feature request #8095:

> Bonus - Allow for text size to be increased automatically if the font is too small for the selected nozzle
> Super awesome bonus - Allow for serial numbers to be added. The text box could ask for a prefix (such as "SN: ") and then a format for the serial number (4 digits max, include/exclude leading 0's, utilize a date, etc), and then how many parts to generate

lukasmatena, 2023-02-22:
Implemented in 2.6.0-alpha2. Closing.

closed 2023-02-22 by lukasmatena