prusa3d/PrusaSlicer #2297: [FR] Comprehensive list of filename variables + fixes

Looxoor, 2019-05-20:
### Version
2.0.0

### Operating system type + version
MacOS 10.14.5

### Request
I can't find anywhere (except the source code referenced in #1680) a comprehensive list of the variables you can use in your filename template. That would be immensely helpful.

I also agree with the poster in #1680 that in a filename a maximum of two decimal places is quite enough.

Great work on the revamped UI etc btw!

bubnikv, 2019-05-21:
It is very much all the parameters that are defined in the print / printer / filament presets plus some more
https://github.com/prusa3d/PrusaSlicer/wiki/Slic3r-Prusa-Edition-Macro-Language

bubnikv, 2019-05-21:
You will find some more information in the release logs, but I agree that this shall be documented. We hope  our content team will now invest some effort into whipping up a PrusaSlicer manual.

bubnikv, 2019-05-21:
From https://github.com/prusa3d/PrusaSlicer/releases/tag/version_1.42.0-alpha1

One may now insert placeholders for the print statistics into the output file name template. This is most useful for the print time estimate, however this Slic3r release supports the following new placeholders: "print_time", "normal_print_time", "silent_print_time", "used_filament", "extruded_volume", "total_cost", "total_weight", "total_wipe_tower_cost", "total_wipe_tower_filament" #714, #922.

bubnikv, 2019-05-21:
I have updated the wiki
https://github.com/prusa3d/PrusaSlicer/wiki/Slic3r-Prusa-Edition-Macro-Language#useful-variables-to-be-used-inside-the-custom-scripts-and-in-the-output-file-name-template

Closing

closed 2019-05-21 by bubnikv