prusa3d/PrusaSlicer #2299: Bed image replaced with graphic glitches

jephstahl, 2019-05-20:
### Version
2.0.0+
PrusaSlicer
Version:   2.0.0+
Build:     PrusaSlicer-2.0.0+-201905201652

Operating System:    Macintosh
System Architecture: 64 bit
System Version:      OS X Mavericks Version 10.9.5 (Build 13F1911)

OpenGL installation
GL version:   2.1 INTEL-8.24.17
Vendor:       Intel Inc.
Renderer:     Intel HD Graphics 3000 OpenGL Engine
GLSL version: 1.20
Installed extensions:

GL_APPLE_aux_depth_stencil
GL_APPLE_client_storage
GL_APPLE_element_array
GL_APPLE_fence
GL_APPLE_float_pixels
GL_APPLE_flush_buffer_range
GL_APPLE_flush_render
GL_APPLE_object_purgeable
GL_APPLE_packed_pixels
GL_APPLE_pixel_buffer
GL_APPLE_rgb_422
GL_APPLE_row_bytes
GL_APPLE_specular_vector
GL_APPLE_texture_range
GL_APPLE_transform_hint
GL_APPLE_vertex_array_object
GL_APPLE_vertex_array_range
GL_APPLE_vertex_point_size
GL_APPLE_vertex_program_evaluators
GL_APPLE_ycbcr_422
GL_ARB_color_buffer_float
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_draw_buffers
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_instanced
GL_ARB_fragment_program
GL_ARB_fragment_program_shadow
GL_ARB_fragment_shader
GL_ARB_framebuffer_object
GL_ARB_framebuffer_sRGB
GL_ARB_half_float_pixel
GL_ARB_half_float_vertex
GL_ARB_instanced_arrays
GL_ARB_multisample
GL_ARB_multitexture
GL_ARB_occlusion_query
GL_ARB_pixel_buffer_object
GL_ARB_point_parameters
GL_ARB_point_sprite
GL_ARB_provoking_vertex
GL_ARB_seamless_cube_map
GL_ARB_shader_objects
GL_ARB_shader_texture_lod
GL_ARB_shading_language_100
GL_ARB_shadow
GL_ARB_sync
GL_ARB_texture_border_clamp
GL_ARB_texture_compression
GL_ARB_texture_compression_rgtc
GL_ARB_texture_cube_map
GL_ARB_texture_env_add
GL_ARB_texture_env_combine
GL_ARB_texture_env_crossbar
GL_ARB_texture_env_dot3
GL_ARB_texture_float
GL_ARB_texture_mirrored_repeat
GL_ARB_texture_non_power_of_two
GL_ARB_texture_rectangle
GL_ARB_texture_rg
GL_ARB_transpose_matrix
GL_ARB_vertex_array_bgra
GL_ARB_vertex_blend
GL_ARB_vertex_buffer_object
GL_ARB_vertex_program
GL_ARB_vertex_shader
GL_ARB_window_pos
GL_ATI_separate_stencil
GL_ATI_texture_env_combine3
GL_ATI_texture_float
GL_ATI_texture_mirror_once
GL_EXT_abgr
GL_EXT_bgra
GL_EXT_blend_color
GL_EXT_blend_equation_separate
GL_EXT_blend_func_separate
GL_EXT_blend_minmax
GL_EXT_blend_subtract
GL_EXT_clip_volume_hint
GL_EXT_debug_label
GL_EXT_debug_marker
GL_EXT_draw_buffers2
GL_EXT_draw_range_elements
GL_EXT_fog_coord
GL_EXT_framebuffer_blit
GL_EXT_framebuffer_multisample
GL_EXT_framebuffer_object
GL_EXT_framebuffer_sRGB
GL_EXT_geometry_shader4
GL_EXT_gpu_program_parameters
GL_EXT_gpu_shader4
GL_EXT_multi_draw_arrays
GL_EXT_packed_depth_stencil
GL_EXT_packed_float
GL_EXT_provoking_vertex
GL_EXT_rescale_normal
GL_EXT_secondary_color
GL_EXT_separate_specular_color
GL_EXT_shadow_funcs
GL_EXT_stencil_two_side
GL_EXT_stencil_wrap
GL_EXT_texture_array
GL_EXT_texture_compression_dxt1
GL_EXT_texture_compression_s3tc
GL_EXT_texture_env_add
GL_EXT_texture_filter_anisotropic
GL_EXT_texture_integer
GL_EXT_texture_lod_bias
GL_EXT_texture_rectangle
GL_EXT_texture_sRGB
GL_EXT_texture_sRGB_decode
GL_EXT_texture_shared_exponent
GL_EXT_timer_query
GL_EXT_transform_feedback
GL_EXT_vertex_array_bgra
GL_IBM_rasterpos_clip
GL_NV_blend_square
GL_NV_conditional_render
GL_NV_depth_clamp
GL_NV_fog_distance
GL_NV_light_max_exponent
GL_NV_texgen_reflection
GL_NV_texture_barrier
GL_SGIS_generate_mipmap
GL_SGIS_texture_edge_clamp
GL_SGIS_texture_lod

### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
Bed image doesn't show up.
It is replaced with a glitchy image that has screen grabs from other apps
[image]
[image]

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_

jephstahl, 2019-05-20:
[image]

jephstahl, 2019-05-20:
[image]

bubnikv, 2019-05-21:
What Mac model is it?

jephstahl, 2019-05-21:
This is a MacBook Pro 13-inch, Early 2011.

bubnikv, 2019-05-21:
I have prepared a release OSX build, where the debug feature of testing every OpenGL call is compiled in. If the error triggers, you should get an error like this in the terminal window, if you manage to execute PrusaSlicer from the terminal:

[2019-05-21 19:33:27.063054] [0x00007280] [error]   OpenGL error in C:\data\src-perl\Slic3r-master\src\slic3r\GUI\GLCanvas3D.cpp:1284, function Slic3r::GUI::GLCanvas3D::init() : 1280 - Invalid Enum

Would you please try the following build?

[PrusaSlicer-2.0.0+4-vb-glcheck-g54474b8-201905211124.dmg.zip](https://github.com/prusa3d/PrusaSlicer/files/3203757/PrusaSlicer-2.0.0%2B4-vb-glcheck-g54474b8-201905211124.dmg.zip)

You will get the console open, if you open the context menu on the application, select "Show Package Content", then go to Content/MacOS/ and double click on PrusaSlicer.

Thanks.

jephstahl, 2019-05-21:
Launched the app via Package Content of the build you provided.
Terminal shows:

> Last login: Tue Feb 19 11:43:26 on ttys001
jephs-computer:~ jephstahl$ /Volumes/PrusaSlicer\ 1/PrusaSlicer.app/Contents/MacOS/PrusaSlicer ; exit;
Multisample enabled

Checked System Console and had these two events:

> 5/21/19 1:45:14.757 PM WindowServer[147]: disable_update_timeout: UI updates were forcibly disabled by application "PrusaSlicer" for over 1.00 seconds. Server has re-enabled them.

> 5/21/19 1:45:17.091 PM WindowServer[147]: common_reenable_update: UI updates were finally reenabled by application "PrusaSlicer" after 3.34 seconds (server forcibly re-enabled them after 1.00 seconds)

didn't see the opengl error you mentioned.

bubnikv, 2019-05-21:
Here is another build, where @enricoturri1966 tried to compress the OpenGL textures. If your system is low on graphics card texture memory, this may or may not help.

[PrusaSlicer-2.0.0+11-et-experiments-gefd247f-201905211227.dmg.zip](https://github.com/prusa3d/PrusaSlicer/files/3203828/PrusaSlicer-2.0.0%2B11-et-experiments-gefd247f-201905211227.dmg.zip)

Thanks.

jephstahl, 2019-05-21:
[image]

That looks better!

What really concerned me was what the corrupted image was  comprised of... I saw bank statements and all kinds of stuff...  I gotta look at a way to blank out the texture memory, if that what it was using.

bubnikv, 2019-05-21:
Cool, thanks for testing.

bubnikv, 2019-08-01:
The upcoming PrusaSlicer 2.1.0-alpha1 implements texture compression, and it loads 8192x8192 textures for OSX Retina and OpenGL 3.0 and newer, otherwise just 4096x4096 textures to conserve GPU memory.

Closing.

closed 2019-08-01 by bubnikv