prusa3d/PrusaSlicer #2300: Feature Request: Hide setting profiles

PRGeno, 2019-05-20:
### Version
2.0 Released

### Operating system type + version
Relevant to all operating systems
 
### 3D printer brand / version + firmware version (if known)
Prusa MK2S & MK3

### Behavior
Feature request to easily hide and show presets, including user, and especially the default system presets.  I can't remember the last time I've used "Verbatim BVOH" filament.  The presetsshow be able to be hidden from the drop down list rather than deleted, and the state not changed even when automatically updated.

#### Project File (.3MF) where problem occurs
Relevant to all models and projects

bubnikv, 2019-05-21:
We don't want to hide all the system presets or make the system presets hideable, but
@vojtechkral is working on an extension of the install wizard to install the filaments, which I suppose will fit your bill, and mine as well.

PRGeno, 2019-05-21:
I'm not sure I understand why the Devs don't want to allow the users to control their own use of the program, but I assume they have their reasons.  

As you indicated, the presets that are really the problem are the Filament presets.  That list is long and filled with noise which frankly makes it pretty useless for most users.   

It is one thing to provide examples and suggestions of how to set up presets for various types of materials. but that list contains many filaments that will never be used by anyone, and some that aren't even available anymore.  

I'm not suggesting adding the ability to delete them, just the ability to hide presets, including the user created presets that a user might not to show normally but doesn't want to delete.

agvol, 2019-05-22:
> I'm not sure I understand why the Devs don't want to allow the users to control their own use of the program, but I assume they have their reasons.
>
> As you indicated, the presets that are really the [...]

As workaround you can delete file from your %APP% path, example file path:
C:\Users\MainUser\AppData\Roaming\PrusaSlicer\vendor\PrusaResearch.ini

After deleting, filament dropdown will show only your items.

bubnikv, 2019-05-23:
> I'm not sure I understand why the Devs don't want to allow the users to control their own use of the program, but I assume they have their reasons.

The reason is our support load. If you are a support personnel, you want to see the same thing on your screen as the customer at the other side of the phone line.

bubnikv, 2019-05-23:
> I'm not suggesting adding the ability to delete them, just the ability to hide presets, including the user created presets that a user might not to show normally but doesn't want to delete.

As noted, we are already working on install wizard extension to install filaments. I suppose that will work for you very well.

torsoreaper, 2019-09-18:
I don't mind all the presets but I do think that my custom ones should be at the top of the drop down, not the bottom.  I wish I could at least just choose the sort order.

PRGeno, 2019-09-18:
> 
> 
> I don't mind all the presets but I do think that my custom ones should be at the top of the drop down, not the bottom. I wish I could at least just choose the sort order.

I prefer having the custom presets at the bottom of the of the list as that puts them closer to where the mouse would likely be.  With the list as cluttered as it is, it's nice to have the ones I use more accessible at the bottom.  So if it's even considered to change the function of the lists, I hope it would be a configurable sort/order option.

rtyr, 2020-06-18:
It is now possible to disable system filament profiles through the configuration wizard so I believe we can close this issue.

closed 2020-06-18 by rtyr