prusa3d/PrusaSlicer #2304: PrusaSlicer 2.0.0.-rc+ g-code error when importing

alesf, 2019-05-21:
### Version
PrusaSlicer 2.0.0.-rc+

### Operating system type + version
macOS 10.14.4

### 3D printer brand / version + firmware version (if known)
Prusa MK3

### Behavior
When I try to load config generated in PrusaSlicer 2.0.0.-rc+ I get an error:
```
PrusaSlicer has encountered an error
Not A Slic3r generated g-code.
```

This also happens if I open the g-code in Slic3r Prusa Edition.

The problem is the first line in g-code:
```; generated by PrusaSlicer 2.0.0-rc+ on 2019-05-17 at 11:04:46```

It works if I change this code to:

```; generated by Slic3r Prusa Edition 1.42.0-beta2+ on 2019-05-17 at 11:04:46```

bubnikv, 2019-05-21:
Fixed with 84965dcb44168421955d8c9ae81d4685b9914753
Thanks for the report.

closed 2019-05-21 by bubnikv

8orge, 2019-05-25:
This i can confirm is an issue .
Went through the same thing just this afternoon .
Was about to post the problem and saw this post :-) .
As long as the first line in the g-code have "Slic3r Prusa Edition" 
it will import that g-code ,so the old slic3r generated g-codes 
is no problem to import in PrusaSlicer 2.0 .