prusa3d/PrusaSlicer #2305: Menu icons on Linux

guestisp, 2019-05-21:
### Version
2.0.0

In many screenshot posted here, i can see some icons in the contextual right menu when right-clicking the object on the plate. (just a random video: https://youtu.be/OK73ApADIfk?t=826 )

On linux there isn't any icon. Is this correct ?

vojtechkral, 2019-05-21:
> On linux there isn't any icon. Is this correct?

Well, sort of. It's the way wxWidgets work. We have asked about this in the wx-users ML and wx devs told us this is a property of the GTK style being used, that it doesn't display bitmaps for (some) buttons and context menus.

So either it's a feature or a bug in wxWidgets. In any case, I don't believe it can be fixed in slicer.

guestisp, 2019-05-21:
Thanks. Just to know.

closed 2019-05-21 by guestisp