prusa3d/PrusaSlicer #2306: Feature Request: Custom preset write protection

lixxbox, 2019-05-21:
### Version
PrusaSlicer-2.0.0+win64-201905201652

### Operating system type + version
Windows (7+10)

### Behavior
All Prusa presets are write-protected and are marked with a lock symbol in the GUI. If changes are to be saved, a new name is suggested with the attachment "- Copy".

Could something similiar be implemented for custom presets?
It is possible to write protect the corresponding ini file in Explorer, this works, but
a) the preset is not marked with a lock symbol and
b) when I try to save the preset no new name is proposed. Instead clicking "ok" without noticing, the changes will be lost (worst case), because the file is write-protected and there is no warning or anything else.

(For me) It would also be sufficient if the file write-protection (Exporer) were taken into account when importing and the preset is then handled like the prusa preset. Maybe a tick in the GUI could be added to protect the current preset?

_Is this a new feature request?_
Yes

jwillmer, 2019-05-21:
I like the idea. Was actually also looking for this feature once in slicer. Basically I made my own preset out of the default presets and would like to write protect it since I always use my preset + small modifications based on the printed object.

rtyr, 2021-01-29:
You can create your own configuration bundle and place it into your configuration folder. It will than behave like standard system profiles. 

See also https://github.com/prusa3d/PrusaSlicer/pull/4122. 

Closing.

closed 2021-01-29 by rtyr