prusa3d/PrusaSlicer #2307: Crash on Export config Bundle (Win64)

pixelk0, 2019-05-21:
### Version
2.0.0+win64 (also 2.0.0 rc2)

### Operating system type + version
Windows 7 64bits

### 3D printer brand / version + firmware version (if known)
Anet A8

### Behavior
- Start Prusa Slicer
- File / Export / Export Config Bundle...
- Choose a file name (same if choosing the default name)
, click on Save
Crashes, exports a partial or empty config bundle

#### Project File (.3MF) where problem occurs
I upload my current config bundle (prior to 2.0 rc2)

[Slic3rPE_A8_Gros3D.zip](https://github.com/prusa3d/PrusaSlicer/files/3202033/Slic3rPE_A8_Gros3D.zip)

pixelk0, 2019-05-21:
If it's any help, the Windows crash report : 
```
Signature du problème :
  Nom d’événement de problème:	APPCRASH
  Nom de l’application:	prusa-slicer.exe
  Version de l’application:	2.0.0.0
  Horodatage de l’application:	5ce2dc98
  Nom du module par défaut:	PrusaSlicer.dll
  Version du module par défaut:	0.0.0.0
  Horodateur du module par défaut:	5ce2dc8c
  Code de l’exception:	c0000005
  Décalage de l’exception:	000000000001deee
  Version du système:	6.1.7601.2.1.0.256.48
  Identificateur de paramètres régionaux:	1036
  Information supplémentaire n° 1:	5f6a
  Information supplémentaire n° 2:	5f6a3082699b63d48b7545df57d672e3
  Information supplémentaire n° 3:	9121
  Information supplémentaire n° 4:	912148daed7b44db5697dbd1c10c5660
```

FidelCapo, 2019-05-21:
Unfortunately, I am unable to reproduce.
Are you exporting bundle to some unusual place?
Can you please zip and attach configuration folder? Help - Show configuration folder.
Thank you.

pixelk0, 2019-05-21:
Here is my config folder (I got it under %appdata%) : [ConfigFolder.zip](https://github.com/prusa3d/PrusaSlicer/files/3202808/ConfigFolder.zip)

I tried to export the config bundle to my D: hard drive into a regular subfolder, as usual and as I did with the previous versions (1.42 beta 2 and before ) successfully.

pixelk0, 2019-05-21:
I just tried from the most neutral start I could Find :
- Installed (French) Windows 7 64 bits into a brand new VirtualBox
- downloaded Prusa Slicer (PrusaSlicer-2.0.0+win64-201905201652.zip) and extracted it in c:\temp
- copied over my 1.42 Config bundle.ini (the one uploaded in the first post)
- Started Prusa Slicer 2 for the first time
- Set it to expert and English(US) (which restarts PS)
- Import the 1.42 bundle
- restart Prusa Slicer (just to be sure)
- Try to export the config bundle => Crash.

bubnikv, 2019-05-21:
I still cannot reproduce it. I tried to load your PrusaSlicer config folder and export, it worked.
I tried to start from scratch as you described, you worked.
I suppose you are running on French windows, maybe that is the reason?

Anybody else was able to reproduce this crash?

pixelk0, 2019-05-21:
I just installed a English Win7 64 bits in another VBOX (just with a French Keyboard). Installed the VBox extensions.
By the way, on the first PS start I cancel the wizard, then import the bundle (didn't change to expert or changed the language (it was already in english this time of course).
Restarted PS, tried to export the bundle and it crashed.
If you want I can upload the virtual machine somewhere (I have more bandwidth tan sanity right now).

pixelk0, 2019-05-21:
If you are adventurous, My virtual Machine (Th Windows isn't activated, but it shouldn't hinder tests) : https://1fichier.com/?8i6v3eygroansy90ko0b
Password : PrusaSlicer

bubnikv, 2019-05-21:
Thanks, that was actually helpful. The config bundle was exported nearly to
the end, and the application crashed when trying to export the names of the
SLA profiles.
These profiles are not set, as you did not install an SLA printer.

void PresetBundle::export_configbundle(const std::string &path, bool
export_system_settings)
is guilty, where
```
    c << "sla_print = " << this->sla_prints.get_selected_preset().name <<
std::endl;
    c << "sla_material = " <<
this->sla_materials.get_selected_preset().name << std::endl;
```
shall be replaced with  `get_selected_preset_name()`, which tests for the
preset index validity. Something for tomorrow.

bubnikv, 2019-05-22:
This should be fixed with f5ef53764069e5ec504dcc17433d80cb83b14d7a
@FidelCapo please retest and close. Thanks.

FidelCapo, 2019-05-22:
This issue should be fixed in next release.
Closing

closed 2019-05-22 by FidelCapo

bubnikv, 2019-05-22:
By the way, for now you can just install the SLA printer, switch the
printer profile to the SLA printer and then import your config bundle.

Sembazuru, 2019-06-27:
I've independently found another workaround for this issue that doesn't require configuring for a printer that I don't have. The configuration bundle is an ASCII text file. I've found that when I save a configuration bundle and Plicer crashes (happens every time on two different Win10 laptops) the last 3 lines are:

```
[presets]
print = 0.20mm QUALITY MK3 (CDE)
sla_prin`
```

If I then delete these last 3 lines (the malformed presets section) then I can import the bundle.

I'm waiting with baited breath for at very least a pre-release of Plicer so I can have this fix.

lt72884, 2019-07-10:
This happens to me all the time on windows 10. Very frustrating. No mastter where i save the .ini file, its either empty area  partial. Here is a small snippet of what happens. thanks

Version	10.0.17763 Build 17763
Processor	Intel(R) Core(TM) i5-3350P CPU @ 3.10GHz, 3101 Mhz, 4 Core(s), 4 Logical Processor(s)

[image]

wildblue, 2019-07-13:
open the file  PrusaSlicer.ini in your config-folder
maybe also check the file  slic3r.ini

change
<
sla_material = 
sla_print = 
>
into
<
sla_material = Prusa Orange Tough 0.05
sla_print = 0.05 Normal
>
open prusa slicer and export your config bundle.

background:
the file vendor/PrusaResearcch.ini is invalid
following must be changed to work on initializing
<
[sla_print:*common*]
compatible_printers_condition = printer_notes=~/.*PRINTER_VENDOR_PRUSA3D.*/ and
>
into
<
[sla_print:*common*]
printer_technology = SLA
compatible_printers_condition = 
>

and
<
[printer:Original Prusa SL1]
printer_technology = SLA
...
>
into
<
[printer:Original Prusa SL1]
inherits = *common*
>
this change results in the correct behaviour, but only if not run once without this fix.
the old config generates the empty config-valurs and these can only be fixed manually at the moment.

hey prusa-team, what is taking so long explaining this or releasing a bugfix? almost2 month since the last release.

don't you have enough developers? i would work for your company, having 10 years experience in c++, c#, digital image recognition, electronics, mechanics. have built my own 3d printer not long ago completely from scratch.

maybe i hear from you, i am located in germany, but could move when nessesary.

with the best regards in this short time

ben

WhiteDragonD, 2019-08-01:
I found a quick fix so you can export before updating to the newer version.... it will trick the program into thinking a SLA profile is installed and won't fail on the export.

Open: **C:\Users\USERNAME\AppData\Roaming\PrusaSlicer\PrusaSlicer.ini**
Under **[presets]** where sla_material and sla_print are empty, add the following:

[presets]
filament = Prusa PET
print = 0.35mm FAST-NAMED
printer = Original Prusa i3 MK2.5-MUSIC
**sla_material = Prusa Orange Tough 0.05
sla_print = 0.05 Normal**

lt72884, 2019-08-01:
Thats interesting that it has to be tricked like that haha. Thanks for
sharing. Now when the profile is imported, do we have to change the SLA
back to pla?

draeath, 2019-08-04:
This remains an issue with PrusaSlicer 2.0.0+win64. It was clearly missed in the release mentioned by @FidelCapo [here](https://github.com/prusa3d/PrusaSlicer/issues/2307#issuecomment-494800342).

Can we get exception handling added around this as a hotfix?

lukasmatena, 2019-08-04:
PrusaSlicer 2.1.0-alpha1 will be released in several days.

draeath, 2019-08-05:
Well, that's not really helpful if I am not interested in testing the other changes in that alpha :P

lukasmatena, 2019-08-06:
Alphas generally precede final releases and I assumed that my statement will be equivalent to saying "we are currently finalizing 2.1.0 release". But clearly that was too implicit :-)

Shortly put: we will do the release after we manage to get master branch stable. Which should be quite soon.

edspeds, 2019-08-11:
Well, I just discovered this bug when I decided to try the 2.1.0 A1.  Of course it didn't automatically import the profiles from 2.0 and when I try to export the configuration bundle so it can be imported into 2.1.0 a1, 2.0 crashes.

WhiteDragonD, 2019-08-11:
> 
> 
> Well, I just discovered this bug when I decided to try the 2.1.0 A1. Of course it didn't automatically import the profiles from 2.0 and when I try to export the configuration bundle so it can be imported into 2.1.0 a1, 2.0 crashes.

Look at my solution above to export. 
https://github.com/prusa3d/PrusaSlicer/issues/2307#issuecomment-517357752

edspeds, 2019-08-12:
> 
> 
> > Well, I just discovered this bug when I decided to try the 2.1.0 A1. Of course it didn't automatically import the profiles from 2.0 and when I try to export the configuration bundle so it can be imported into 2.1.0 a1, 2.0 crashes.
> 
> Look at my solution above to export.
> [#2307 (comment)](https://github.com/prusa3d/PrusaSlicer/issues/2307#issuecomment-517357752)

That did it, thanks!  Sorry it took a couple of days to put this to the test.