prusa3d/PrusaSlicer #2308: Thick wall vase mode

espr14, 2019-05-21:
v2.0.0 win64

Vase mode has these advantages:
1. No seems
2. Faster print (no jumps)
3. No retractions (no oozing, good for flex)
4. Water tight

The main disadvantage is thin walls. But you can make the walls much stronger by adding a seem and still printing in single continuous line. Watch this model [https://www.thingiverse.com/thing:2277253](https://www.thingiverse.com/thing:2277253). The seem looks much better than an ordinary seem. This model has infill and inner wall that make it really strong.

We can go further and create a model which has infill, inner and outer wall with a single seem. You could even change type of infill (rectilinear, zig-zag, ...) and density! Following picture shows inner and outer wall with arrows pointing to area where infill can be.

[image]

Now we can add another features:
1. Continuous bottom infill
2. Continuous top infill

There are lots of single line filling patterns that can be used. Vase mode need not to eliminate top infill. Z height can be stopped and afterwards it can continue in an ordinary smooth Z changes.

With these enhancements you can print much more models with much stronger walls. Similar (simpler) proposal was here already (#399). I suppose this will be closed as well but maybe it would be implemented in ten years time and used heavily ever after.

espr14, 2019-10-23:
This vase mode with infill can be used for multimaterial prints while purge into infill is turned on. However, this requires to purge into infill only (without using wipe tower at all).

Panayiotis-git, 2019-10-23:
I think that even purge into infill is selected, the wipe tower is still required for the filament swapping.

espr14, 2019-10-23:
Currently, the wipe tower is a must. However, you can change the code so the wipe tower will be not used at all. Then the infill density should change accordingly to color change. Moreover, user has to be aware that the color transition could be visible. But in some cases some slight color transition need not to matter.

bubnikv, 2020-02-08:
Please provide a pull request.

closed 2020-02-08 by bubnikv