prusa3d/PrusaSlicer #2311: Material extruding while print head is returning to print position after a color change

plamhut, 2019-05-21:
### Version
PrusaSlicer 2.0.0

### Operating system type + version
MAC OS 10.11.6

### 3D printer brand / version + firmware version (if known)
MK3S Firmware 2.7.1

### Behavior
Material extruding while print head is returning to print position after a color change.
Create a project with a color change at about 25% down from the top layer. (Position not critical)  Print the created Gcode. Normal color change process happens. After answering yes to is the color correct the print head goes back to the correct position to continue printing, but it is extruding material on the way back. As long as I hold the extruded material that was created by the color change so I can pull it off after the move instead of before the move as a work around printing resumes properly. 

Bug fix not a new feature.

#### Project File (.3MF) where problem occurs

[Barron Favor18x25.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/3203170/Barron.Favor18x25.3mf.zip)
[Barron Favor18x25.stl.zip](https://github.com/prusa3d/PrusaSlicer/files/3203181/Barron.Favor18x25.stl.zip)

bubnikv, 2019-05-21:
This is a feature request to be reported here:
https://github.com/prusa3d/prusa-firmware/issues

But AFAIK it works as intended, it is there to pressurize the nozzle. I
personally just watch for it and pull the excess filament with the tweezers.

bubnikv, 2019-05-21:
Closing, as it is a firmware issue / request.

closed 2019-05-21 by bubnikv